OpenEVSE/openevse_esp32_firmware #5: Move timer functions to WiFi module to remove the need for RTC hardware

glynhudson, 2019-08-16:
WiFi to control sleep and charging with RAPI commands while on timers and eliminate the RTC chip. The chip and parts add significant cost to every batch of LCDs we make. Also timer functions take up considerable code space on the open_evse controller.

Time could be updated automatically via NTP and held in memory.

glynhudson, 2019-10-04:
I think the ESP32 has an internal RTC we can use? Jjust need to add a small external battery?

jeremypoulter, 2019-10-05:
I am afraid I don't think it is that simple, the RTC is to support the ESP32's deep sleep, so (for our use case) you would need some external hardware to give a signal to the ESP32 when the main power is connected or not as well as switch between being powered by battery or not.

When the external power is disconnected the battery would take over powering the ESP32 and the ESP32 would then need to switch to deep sleep mode. Especially if being powered by a CR2032 we would have to be really quick at that switch off not that that is too bad, can probably just be a bit brutal and switch off in the interrupt.

Perhaps a nicer solution would be to use a small LiPo battery in a more of a backup battery style, ie do a 'graceful' shutdown before going in to low power mode. The [Adafruit HUZZAH32](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather) would be a good candidate to try that on, although I do not [see](https://cdn-learn.adafruit.com/assets/assets/000/041/630/original/feather_schem.png?1494449413) a good method of detecting the power source, maybe wiring the charger STAT pin to a free GPIO (and disconnect the charge LED)?

jeremypoulter, 2019-10-05:
You may also need an external crystal: https://www.esp32.com/viewtopic.php?t=3715, that may or may not be on the ESP32 module.

chris1howell, 2019-10-05:
For the vast majority, timer support via NTP sourced clock will be
sufficient, RTC is redundant (and expensive).

For those with an need for a clock while offline adding an external RTC is
not difficult. There are pleanty avaliable.

commit 18df3f70 (2021-01-24): Move timer functions to WiFi module to remove the need for RTC hardware
Fixes #5
closed 2021-03-07 by jeremypoulter