OpenEVSE/openevse_esp32_firmware #7: WiFi Scan Improvements

glynhudson, 2019-08-16:
In order of priority:

- Don't scan until the user clicks a 'Scan' button, this will avoid scanning for networks if a user wants to continue in standalone mode 
- Only scan once, this will avoid the list changing in size while the user is trying to enter their passkey
- Sort SSID'S by RSSI in descending order (less negative RSSI = stronger signal = display first)
- Don't display very weak networks 

These seemingly minor tweaks are actually very important, it took many attempts to setup a WiFi in a location with 100's of WiFi networks since the wifi network scan table kept jumping around in size while trying to enter a complex PSK. Most of the WiFi networks where very weak and should not have been displayed. Also if a user wants use to use stand-alone more it's frustrating to have to wait for a wifi scan that is not needed.

glynhudson, 2019-10-01:
As a backup we could have a direct text entry field for SSID and PSK, this would also allow connection to hidden SSIDs.

jeremypoulter, 2019-10-01:
Discussed some additional ideas on this. A slightly nicer solution maybe:

- Limit length of the SSID table to 5 and make scrollable? 
- Merge duplicate SSIDs, display lowest RSSI
- Always show the selected SSID
- Probably implement #32 as well

jeremypoulter, 2019-10-14:
https://eyesaas.com/wi-fi-signal-strength/

glynhudson, 2019-10-14:
Maybe we should hide all networks with less than -70 or -79 dBm

jeremypoulter, 2019-10-15:
I was going to use that page to give a strength 0-5 based on those values with 0 being >= -80 and probably hide those, then use some nice icons for 1-5.

glynhudson, 2019-10-15:
Cool, that sounds reasonable

jeremypoulter, 2019-10-21:
I think this is largely complete now, give it a try. Should also work on ESP8622 version

glynhudson, 2019-10-24:
I've just tested this, few minor thoughts and issues:

- The new UI is really nice and the scanning seems to work better, the list doesn't just around as much 
- The signal strength indicator does not seem to work for some networks e.g our network `OpenEnergyMonitor` was shown as not having any signal bars even though the ESP is located right next to the router. 
- When a network is selected the SSID is not populated in the SSID text input field, this results in an error "Please Select a Network" when trying to connect. 

See screenshots:

[image]

[image]

OpenEnergyMonitor network should have full signal strength in this location:

[image]

jeremypoulter, 2019-10-25:
Fixed the selection issue, was missing a bit of code from the AP page. 

For the strength can you send me the content of http://192.168.4.1/scan  ?

glynhudson, 2019-10-30:
I've just tested the latest version, the WiFi scan in the Captive Portal still does not seem to populate the SSID when I click on a network.  However, when I enter "Stand alone WiFi mode" then click a network name the SSID does get populated. Therefore, the issue only effects the initial captive portal WiFi scan page. 

The signal strength seems to be working, need to do some more testing in a location with more networks! 

Another issue I've noticed if once I'm connected to a network if I click "Change WiFi network" then "Scanning..." appears but network list does not get populated `/scan` returns `no properties` :

[image]

jeremypoulter, 2019-11-03:
The WiFi portal may have been that I had not pushed the updated header files. Done that now and tested locally, looks to work.

I can replicate the scan issue in client mode, do not think this is related to these changes as I these where only GUI changes. It is also probably not massively important as you can always factory reset to connect to a different WiFi network. Maybe it is best to open as a separate issue?

closed 2019-11-05 by glynhudson