OpenEVSE/openevse_esp32_firmware #8: Add Energy Savings

chris1howell, 2019-08-16:
Add basic energy savings.

Add ability to turn of LCD back light after x minutes of inactivity with the RAPI command $FB 0.

back light will turn back on automatically with any state change.

brettski74, 2020-11-09:
While I doubt the energy savings will be much - probably not more than a few cents a month. The bigger issue is a brightly lit screen drawing attention to an outdoor installation at night.

temperodave, 2020-11-10:
I agree the LED off while not actively charging would be nice to reduce light pollution and drawing attention to the installation.

Tifaifai, 2022-01-13:
I also agree the LED off. On the other hand, as for the smoke detectors, two LCD or LED flashing per minute will allow to know the status of the openevse when the LCD is off. (No button to activate the openevse)

zorkerz, 2022-07-25:
Very much want this feature as well. Currently I unplug my openevse and stop charging to watch stars at night.

alandpearson, 2023-03-27:
Hi @jeremypoulter 

I've wanted this feature for quite a while, but see it has been open for 4 years.
Is this complex to add ? You've already added the dimming feature, and I was thinking if dim could go to zero (backlight off) that would be a good enough for now and the automatic backlight-on-status-change is a nice to have.

Thank you
Alan

jeremypoulter, 2023-03-27:
The dimming is for the LEDs, but we can however set the backlight off. The problem is more when to turn it back on, especially for units with no button. 

That being said this is probably going to be the next issue to work on.

alandpearson, 2023-03-27:
Thanks Jermey,  I was wondering if I could help, but I think its probably too early for me until I get to know the codebase and how it works.
I’ll stick with trying to turn a GPIO on for relay control for now :)

alandpearson, 2023-04-02:
Hi Jermey,

I've create a pull request https://github.com/OpenEVSE/open_evse/pull/16  for you to have a look at.
I think it should work, and implement a timeout for the backlight.
I'm not able to test it though (I only have a production EVSE) and it does need some work.

It would probably make sense to be able to disable this, by say setting the timeout to 0.

From my simplistic analysing of the code, some things are obvious
1) The backlight setting / colour appears to be refreshed every time the LCD is, necessitating remembering the last time its state was changed (not refreshed)
2) J1772 would appear to set the backlight state every time there is a state change, which would appear to do what we want ?

Let me know what you think.

jeremypoulter, 2023-04-02:
I will take a look, but probably it would be better for the LCD backlight to be controlled by the WiFi firmware, that is where the majority of the logic is now. One thing to bear in mind is that the backlight should remain on in error conditions.

alandpearson, 2023-04-02:
I think there is a lot of confusion between the LED and LCD.  Looks like J1772 seems to control it for the most part, which would seem sensible to me (it knows what condition its in and backlight reflects that)>
I think it will stay on in error conditions as the update logic essentially seems to stop doing other updates in an error. Needs tested.

jeremypoulter, 2023-04-02:
Definitely a lot of confusion between the LCD and LED, but not in this case ;-) (although the LED probably does need to do the same) the vast majority of the LCD control at this point comes from the WiFi module via RAPI commands and probably will move the LCD to be directly connected to the WiFi module at some point. It also is a lot harder to update the EVSE firmware we try not to implement new features on the EVSE board.

alandpearson, 2023-04-02:
That's  a good point on the difficulty of updating it.  It would be great if ESP32 could flash the AVR (like esplink) but the amount of code and complexity to make that so would be daunting, and likely not even have the room to do so.

I had in an other thread mooted the idea of moving the LCD control to the wifi board (to free up the i2c port), but looking at the code, it is a seriously heavy lift. I'm not sure if the EVSE can work without the wifi board, so having the LCD on openevse may need to stay if that is the case.  I've no skin in the game, I just want the damn backlight off  :)

FYI...  because I haven't quite learnt the etiquette of branching quite yet, this pull request also includes the code to allow relays to work independently...  you'll see in the other thread you're tagged in.

adobe75, 2025-01-19:
Hello, I would like to have the option to disable the LCD backlight in my OpenEVSE. Is this feature still planned?

root0x, 2026-09-15:
Commenting just to keep this issue alive, would be great to have the ability to turn the backlight off

RAR, 2026-09-21:
Implemented in #1286 (firmware — backlight off after `lcd_backlight_timeout`, back on for any state change, button press, charging or fault) with the matching UI control in OpenEVSE/openevse-gui-nightshift#142. Both are **untested on hardware** — if anyone with a 2-line LCD unit (RGB or mono backlight) can try the PR build, the test checklist is in the PR.

closed 2026-10-03 by chris1howell