OpenEVSE/openevse_esp32_firmware #12: Wired etherent connectivity

glynhudson, 2019-08-21:
Test using Olimex ES32-gateway with Ethernet to be used instead of Wi-Fi when installation is more suited to wired Ethernet 

https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware

jeremypoulter, 2019-08-27:
Added

glynhudson, 2019-09-24:
Ethernet dev board pinout connections:

[image]

glynhudson, 2019-09-30:
I've just tried to test Ethernet on the Olimex ES32-gateway board, however when I try to compile the `openevse_esp32-gateway` environment I get `Error: The program size (1312862 bytes) is greater than maximum allowed (1310720 bytes)`

Have you able to compile the current version for the ES32 gateway board?

jeremypoulter, 2019-09-30:
Travis CI also showed this issue for some reason, oddly worked fine locally. In any case been meaning to look at the binary size because building the debugger build fails for the same reason. 

Given we are not using SPIFFS I opted to just use a different [flash partition table](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables) to give us an extra 40% space.

Not tried it but I suspect this will not update over OTA.

glynhudson, 2019-10-01:
Thanks, it compiles fine now. 

I've just tested updating Huzzah32 via HTTP update via web interface with a .bin compiled with `min_spiffs.csv` and it worked fine!

glynhudson, 2019-10-17:
I've finally managed to get this up and running, it turns out the silkscreen on my hardware rev C is different to yours! Here are the location of Rx, Tx: 

[image]

Unfortunately the Ethernet does not seem to be working, I get these errors in the debug serial: 

```
E (2094) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
E (3094) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
E (3094) emac: Initialise PHY device Timeout
[E][ETH.cpp:101] begin(): esp_eth_enable error: -1
```

jeremypoulter, 2019-10-17:
Is a Rev.E on that picture ;-) can you try 
[platformio_esp32_eth.zip](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/files/3740844/platformio_esp32_eth.zip) ?
If that does not work, can you try the WiFi->ETH_LAN870 examples from the Arduinio IDE

jeremypoulter, 2019-10-18:
Ah here is the issue I think: https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/SOFTWARE/Newest-board-revisions/ETH_PHY_demo/main/ethernet_example_main.c#L59

That logic for the power pin is not in the rev B/C sample: https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/SOFTWARE/Hardware-revisions-B-or-C/ETH_PHY_demo/main/ethernet_example_main.c

jeremypoulter, 2019-10-21:
The Arduino core has a setting for the revision of the board, in Platform IO this is done in `platform.ini` using:

```
board_build.extra_flags = "-DARDUINO_ESP32_GATEWAY=\'E\'"
```

In addition to this there is also (now) a conflict with Serial2 and the Ethernet... GPIO17 is used for both so moved `TX2` to GPIO 32 so the pinout is also changed.

jeremypoulter, 2019-10-21:
[image]
[image]

closed 2019-10-21 by jeremypoulter

jeremypoulter, 2019-10-21:
Fixed in ef4995d4db911245ba43134ac5e803627eb07341

glynhudson, 2019-11-01:
Awesome, just tested this and it's working :-) 

```
Got Network event SYSTEM_EVENT_ETH_CONNECTED
ETH Connected
Free: 215164
Got Network event SYSTEM_EVENT_ETH_GOT_IP
ETH MAC: 24:0A:C4:26:85:03, IPv4: 192.168.0.102, FULL_DUPLEX, 100Mbps
Connected, IP: 192.168.0.102
```

commit 3df2f978 (2019-09-30): As we are not using SPIFFS use the `min_spiffs.csv` partition table

Fix #12

See https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables for more info