OpenEVSE/openevse_esp32_firmware #18: Emoncms http posting incorrectly reports "connected: no"

glynhudson, 2019-09-07:
Posting to Emoncms (emonPi) via http / https works great but even when successfully posting the interface still reports `Connected: No` and debug log reports `Emoncms error: {"
success": true}`

[image]

[image]

Currently, I'm posting to an emonPi which does not support https, is https only enabled for emoncms.org and data.openevse?

jeremypoulter, 2019-09-09:
The problem here is EmonCMS behaves differently locally vs emoncms.org;

'Post' to emoncms.org:
```
HTTP/1.1 200 OK
Date: Mon, 09 Sep 2019 09:48:54 GMT
Server: Apache/2.2.22 (Ubuntu)
X-Powered-By: PHP/5.3.10-1ubuntu3.25
Content-Length: 2
Connection: close
Content-Type: application/json

ok
```

Post to local EmonBase:
```
HTTP/1.1 200 OK
Date: Mon, 09 Sep 2019 09:50:11 GMT
Server: Apache/2.4.10 (Raspbian)
Content-Length: 17
Connection: close
Content-Type: text/plain;charset=UTF-8

{
  "success": true
}
```

If you can confirm that the HTTP code is 2xx if successful and 4xx/5xx on all failures in both cases then I can change to use the HTTP return code to dictate success. If not will have to try and parse both cases. Looking at the Node.js version that is what happens there.

Note: the `Content-Type` is wrong in both cases.

glynhudson, 2019-09-15:
I've raised this issue with @TrystanLea, it's because emoncms.org is running a different (older) version than the local Emoncms. 

Would it be possible to handle either `ok` or `"success": true` as a positive result? This would work for all emoncms installs. This is probably more reliable than just looking for 2xx HTTP code.

commit 80e0955d (2019-11-04): Fixed posting to Emon Pi

Fixes #18
closed 2019-11-04 by jeremypoulter