OpenEVSE/openevse_esp32_firmware #19: Hostname lookup not working

glynhudson, 2019-09-07:
Hostname lookup e.g htp://openevse does not seem to work on the ESP32

glynhudson, 2019-09-17:
Ah http://openevse.local/ is now working for me. 

However, this is soon to change to http://openevse-<ChipID> see https://github.com/OpenEVSE/ESP32_WiFi_V3.x/issues/15

glynhudson, 2019-09-19:
Actually, hostname lookup (http://openevse.local) is working for me on desktop browser but not on mobile (Android).

jeremypoulter, 2019-09-20:
Android does not support .local (mDNS) that I am aware of.

glynhudson, 2019-09-20:
http://openevse hostname used to work on both my laptop and android
smartphone on the esp8266 version.

- sent from my mobile device

jeremypoulter, 2019-09-23:
There are two different cases here:

1. http://openevse - This will use normal DNS to lookup the IP address based on your default domain (or maybe setup in your `/etc/hosts` file). Some routers will take the hostname sent via DHCP options and set that in the local router DNS. My router (OpenWRT base) does this so http://openwrt and http://openwrt.lan both work. Because this is just 'normal' DNS it will work on anything
2. http://openevse.local - As mentioned above this depends on mDNS support and not supported on some systems (https://android.stackexchange.com/questions/49188/how-to-get-mdns-working-for-chrome-on-android), should work on Apple stuff, newer desktop Linux and Windows (with iTunes installed) 

Can you check your router to see what (if any) hostname the OpenEVSE is sending? There is normally a page that lists the IP address and name of any connected device.

glynhudson, 2019-09-23:
My router (Google WiFi) is reporting the correct hostname. Strangely on the esp8266 V2 version http://openevse.local didn't work while http://openevse did! 

[image]

I'm not fussed about `.local` working, but the DNS lookup was very useful, especially since this worked on android. Currently, the .local lookup works on my desktop browser and ipad.

glynhudson, 2019-12-02:
I've just noticed that http://espressif/ hostname works on Android. It would be great if this hostname could be the same as the mDNS hostname (.local) which only works on desktop machines. Android http://openevse hostname lookup used to work on the ESP8266 version on Android and doesn't on the ESP32 vesion. 

Can you see where http://espressif is set? 

[image]

glynhudson, 2019-12-03:
Just tested again and hostname lookup seems to be working at least on my pixel 3 

[image]

glynhudson, 2019-12-07:
Could be related to: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/249

jeremypoulter, 2019-12-10:
I added potential fix and some additional debugging. @glynhudson please can you give it a try and let me know if it works and if not send me the first bit of the log looking like:

```
Connecting to SSID: wibble
Got Network event SYSTEM_EVENT_WIFI_READY
Got Network event SYSTEM_EVENT_STA_START
WiFi.getHostname() = openevse-a7d4
After net_setup: 254104
Server started
After web_server_setup: 250652
After ota_setup: 248244
last_state = -1
state = 0
divertmode = 1
Got Network event SYSTEM_EVENT_STA_CONNECTED
Connected to wibble
Got Network event SYSTEM_EVENT_STA_GOT_IP
Connected, IP: 172.16.0.205
WiFi.getHostname() = openevse-a7d4
```

glynhudson, 2019-12-10:
Just tested, I seem to be still getting `WiFi.getHostname() = espressif`

```
Got Network event SYSTEM_EVENT_STA_START
After net_setup: 252108
Server started
After web_server_setup: 248636
After ota_setup: 246212
last_state = -1
state = 0
divertmode = 1
val = 3
state = 3
last_state = 0
state = 3
divertmode = 1
first read RAPI values
Got Network event SYSTEM_EVENT_STA_CONNECTED
Connected to Crumble
Got Network event SYSTEM_EVENT_STA_GOT_IP
Connected, IP: 192.168.86.43
WiFi.getHostname() = espressif
MQTT Connecting to...emonpi
lastWill = {"state":"disconnected","id":"c44f330d55ad","name":"openevse-55ad"}
MQTT connected
```

jeremypoulter, 2019-12-11:
Looks like this may be an ESP32 core issue espressif/arduino-esp32#2537

jeremypoulter, 2019-12-12:
After a bit more testing looks to be resolved in the latest IDF and I have verified this with the `openevse_esp-wrover-kit_idf` env.

jeremypoulter, 2019-12-13:
The IDF build breaks RAPI comms so I reverted the Huzzah32 build to just Arduino and added the `openevse_huzzah32_idf` to build under IDF/Arduino. @glynhudson Please give that a try and see if it resolves the Host name issue for you

glynhudson, 2019-12-14:
Yes, compiling using `openevse_huzzah32_idf` fixed the hostname issue http://openevse-xxx/ now works on android and my desktop browser! 👍 

I've not noticed any RAPI issues yet, although I've not tested much yet. What exactly has broken?

jeremypoulter, 2019-12-15:
For me calling the RAPI endpoint was returning a 500 error and 'unknown'  I will take a closer look.

glynhudson, 2019-12-16:
You're correct, when running `openevse_huzzah32_idf` even though the main functions work lots of RAPI calls are failing:
 
[image]

glynhudson, 2019-12-17:
The fix for this is currently implemented in IDF 3.3, currently arduino-esp32 is 3.2. We will wait for arduino-esp32 to updater to 3.3 since compiling under IDF breaks functionality.  

https://github.com/espressif/arduino-esp32/releases

jeremypoulter, 2020-01-23:
Sadly the update of the OSF for Arduino did not fix the issue, see my comments [here](https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-577608379)  :(  However I have fixed the few issues I was aware of in the IDF build.

@glynhudson can you give the `openevse_huzzah32_idf` build a try again?

glynhudson, 2020-01-27:
I've just tested `openevse_huzzah32_idf` and it seems to working well. I'm no longer getting RAPI failures as before:

[image]

Hostname lookup seems to be working, but more testing on different networks / devices is needed to ensure this isn't just due to the local DNS cache on my router! 

Good effort trying to trace the cause of this bug in the espressif code. 

Is using `openevse_huzzah32_idf` the best solution for the moment? What are the pros and cons of this?

glynhudson, 2020-02-04:
Fixed using `openevse_huzzah32_idf` environment

commit f807c39c (2019-12-12): Use the IDF to build the Huzzah32 version for testing if the latest IDF fixes #19
closed 2020-02-04 by glynhudson