OpenEVSE/openevse_esp32_firmware #20: Setting the time does not work

jeremypoulter, 2019-09-09:
Clicking the update time button sends the RAPI command but the connection is immediately closed.

glynhudson, 2019-09-09:
Setting the time seems to work fine for me using Chrome

glynhudson, 2019-09-26:
Seen updating I seem to be having issues setting the timer, see video: https://drive.google.com/file/d/13mJm9XhdJiivJOPdcbcGyOC0waMSQUp2/view

Are you able to replicate?

jeremypoulter, 2019-09-27:
ArduinoMongoose library?

glynhudson, 2019-09-27:
Do you mean I need to update the ArduinoMongoose library? Should this happen during the platformio compile process or is this a manual step?

jeremypoulter, 2019-09-27:
May need to force it as it is not properly managed yet

glynhudson, 2019-09-27:
No worries, I've done `pio lib install https://github.com/jeremypoulter/arduinoMongoose.git`. It seemed to pull in some new changes. Will report back once I've updated. I'm still having trouble with OTA so will need to open up the box to do a physical update. Thanks.

jeremypoulter, 2019-09-27:
For OTA try
```
curl -F 'file=@.pio/build/openevse_huzzah32/firmware.bin'  http://172.16.0.157/update && echo
```
(obviously change the IP)

glynhudson, 2019-09-28:
Fantastic! That worked, so much easier than a physical serial upload. Time setting is now working great with the latest mongoose lib 👍

closed 2019-09-28 by glynhudson