OpenEVSE/openevse_esp32_firmware #21: Memory leak

glynhudson, 2019-09-09:
The free heap seems to be reducing linearly! After being running for about 24hrs the unit required a rest to become responsive. Since then, I've setup better logging to capture the freeheap to see what value resulted in a crash. I will continue to monitor 

[image]

glynhudson, 2019-09-15:
It's now been 6 days and the ESP32 is still up and running, the free heap seems to nosedive then recover on an almost periodic basis:

[image]

As a comparison here is a graph of the freeheap of the ESP8266 version over a 7 day period:

[image]

glynhudson, 2019-11-01:
After the latest update (WiFi scan) the ESP32 has been very unstable, it only runs for 10hrs before hanging. A physical reset is then required to bring the module back to life. When the unit is in the hanging state everything including serial debug, MQTT, HTTP and Emoncms posting stops. 

This shows the graph of free heap, unlike previously the unit does not automatically reboot. I have been manually rebooting ever 24hrs or so. This is running on a Huzzah 32 

[image]

CircuitSetup, 2019-11-06:
I was having a very similar problem with my version of EmonCMS, which was modified to use ESP32. I found the instability and freezing would happen more quickly when the GUI was left open. In my case the memory leak was caused by the async library not releasing sessions (espasyncwebserver, and AsyncTCP). I see that this is using a different library for the webserver, but it may be something to look into.

glynhudson, 2019-11-14:
Switching off HTTP posting to Emoncms has made a significant differance. The unit has not frozen since and the free ram does not go down so low. However, it looks like the memory leak is still preset: 

[image]

glynhudson, 2019-12-02:
Fixed by https://github.com/OpenEVSE/ESP32_WiFi_V3.x/commit/451db125602c7b0b1cd3462400d9695190626848

[image]

closed 2019-12-02 by glynhudson