OpenEVSE/openevse_esp32_firmware #24: Hostname XMLHttpRequest blocked by CORS policy

glynhudson, 2019-09-22:
Using hostname lookup in a desktop browser e.g http://openevse.local results in the following errors. These errors result in some parameters not being displayed e.g max current

`Access to XMLHttpRequest at 'http://192.168.86.20/r?json=1&rapi=$GH' from origin 'http://openevse.local' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.`

## Using hostname access:

[image]

## Using direct IP address (no errors):

[image]

jeremypoulter, 2019-09-23:
This will be also related to setting headers not working jeremypoulter/ArduinoMongoose#3

jeremypoulter, 2019-09-29:
Fixed with jeremypoulter/ArduinoMongoose#3

glynhudson, 2019-10-01:
I'm still experiencing the same issue when using hostname lookup after updating 

[image]

jeremypoulter, 2019-10-01:
Are you using the latest ArduinoMongoose library? Probably a good time to register it properly...

jeremypoulter, 2019-10-01:
Just pushed an update to force the latest version

glynhudson, 2019-10-01:
Yes, I'm pretty sure I'm running the latest version. Before compiling I ran `pio lib install https://github.com/jeremypoulter/ArduinoMongoose.git`. Just tested again with same result

glynhudson, 2019-10-01:
Ok, fixed now using V0.0.3 of Arduino Mongoose

closed 2019-10-01 by glynhudson