OpenEVSE/openevse_esp32_firmware #25: HTTP update not working

glynhudson, 2019-09-22:
Trying to update via HTTP web interface results in ESP32 rebooting but update is no applied. The firmware file does not seem to uploading correctly. Browser does not register a file upload: 

[image]

Here is the debug log showing the update process starting them immediately rebooting 

```
Free: 250504
Free: 251192
Free: 251652
Free: 251652
Free: 251652
Update Start: firmware.bin
Update: 0%

0:32
Got 'emonpi' 144 == 144 @ 282:32
Got 'emonpimqtt2016' 137 == 137 @ 314:64
Got 'emon/emontx3/power2' 242 == 242 @ 378:30
Got '' 128 == 128 @ 408:30
Got '' 255 == 128 @ 498:15
Using default ''
```

jeremypoulter, 2019-09-23:
Hum... Will check on this. Seen something similar when selecting update with no file...

glynhudson, 2019-09-24:
Could be related toi https://github.com/OpenEVSE/openevse_wifi_gui/pull/16

glynhudson, 2019-09-28:
As a work around firmware can be updated using 

`curl -F 'file=@firmware.bin' http://<IP-ADDRESS>/update && echo`

As mentioned in https://github.com/OpenEVSE/ESP32_WiFi_V3.x/issues/20

jeremypoulter, 2019-09-28:
Good to know, that confirms what I suspected. It is an issue with the new UI changes OpenEVES/openwifi_server_gui#16

glynhudson, 2019-09-30:
Awesome! HTTP update is working great, the animation is really nice 👍 

[image]

glynhudson, 2019-10-21:
I've just tested HTTP update again and it seems to be uploading correctly but reboot does not happen: 

```
Update Start: firmware.bin
Update: 0%
Update: 1%
Update: 2%
Free: 241748
Update: 3%
Update: 4%
Update: 5%
Update: 6%
Update: 7%
Update: 8%
Update: 9%
Update: 10%
Free: 245020
Update: 11%
Update: 12%
Update: 13%
Update: 14%
Update: 15%
Update: 16%
Update: 17%
Update: 18%
Update: 19%
Free: 242268
Update: 20%
Update: 21%
Update: 22%
Time1
Update: 23%
Update: 24%
Update: 25%
Update: 26%
Update: 27%
Update: 28%
Free: 242136
Update: 29%
Update: 30%
Update: 31%
Update: 32%
Update: 33%
Update: 34%
Update: 35%
Update: 36%
Update: 37%
Free: 243680
Update: 38%
Update: 39%
Update: 40%
Update: 41%
Update: 42%
Update: 43%
Update: 44%
Update: 45%
Free: 240384
Update: 46%
Update: 47%
Update: 48%
Update: 49%
Update: 50%
Update: 51%
Update: 52%
Update: 53%
Update: 54%
Free: 245484
Update: 55%
Update: 56%
Update: 57%
Update: 58%
Update: 59%
Update: 60%
Update: 61%
Update: 62%
Update: 63%
Update: 64%
Free: 240576
Update: 65%
Update: 66%
Update: 67%
Update: 68%
Update: 69%
Update: 70%
Update: 71%
Update: 72%
Update: 73%
Update: 74%
Free: 246028
Update: 75%
Update: 76%
Update: 77%
Update: 78%
Update: 79%
Update: 80%
Update: 81%
Update: 82%
Update: 83%
Free: 246080
Update: 84%
Update: 85%
Update: 86%
Update: 87%
Update: 88%
Update: 89%
Update: 90%
Update: 91%
Update: 92%
Update: 93%
Update: 94%
Free: 242264
Update: 95%
Update: 96%
Update: 97%
Update: 98%
Update: 99%
Free: 252240
Free: 252240
Free: 251776
Free: 252444
Free: 252444
Free: 251776
Free: 251644
Time1
Free: 251776
```

Triggering a manual reboot reslts in dataloss and all settings being returned to factory default: 

```
ng default ''
Got '� � ' 32 == 200 @ 32:64
Using default ''
Got '�)' 130 == 11 @ 542:33
Using default ''
Got 'Vh� ' 52 == 83 @ 128:45
Using default 'data.openevse.com/emoncms'
Got '��� � ' 145 == 254 @ 173:32
Using default 'openevse-55ad'
Got '
=�=�=�=�� ' 137 == 106 @ 438:60
Using default ''
Got '#�� ' 137 == 244 @ 205:45
Using default 'emonpi'
Got ' � ' 11 == 64 @ 250:32
Using default 'openevse-55ad'
Got '���t�� � ' 128 == 164 @ 282:32
Using default 'emonpi'
Got '������ � ' 153 == 89 @ 314:64
Using default 'emonpimqtt2016'
Got '=�=�=�=���� ' 240 == 186 @ 378:30
Using default ''
Got '=�=�=��� ' 137 == 85 @ 408:30
Using default 'emon/emonpi/power1'
Got '	=�=�=�=��Ŷ�' 1 == 241 @ 498:15
Using default ''
Got '�8� ' 32 == 132 @ 513:15
Using default ''
Got '' 208 == 128 @ 528:10
Using default ''
Got 'b6c020'  0 == 214 @ 538:4
Using default '000000'
After config_load_settings: 308876
Configuring pin 13 for LED
Starting AP
Got Network event SYSTEM_EVENT_WIFI_READY
Got Network event SYSTEM_EVENT_AP_START
Got Network event SYSTEM_EVENT_AP_START
Got Network event SYSTEM_EVENT_AP_START
AP IP Address: 192.168.4.1
After net_setup: 260876
Server started
After web_server_setup: 257624
After ota_setup: 255136
last_state = -1
state = 0
divertmode = 1
val = 254
state = 254
last_state = 0
state = 254
divertmode = 1
first read RAPI values
Free: 254796
Free: 255252
Free: 255252
Free: 255252
Free: 255252
```

glynhudson, 2019-12-03:
Last time I used HTTP update via the web interface to update from 3.0.0 to 3.0.1 the update it worked fine. However, all the EEPROM config was lost and the unit reverted to AP mode on factory settings.

glynhudson, 2019-12-10:
HTTP update now seems to be working well on the Huzzah, however it fails every time on the **ESP Gateway (rev E)** connected via Ethernet. I've not tried it connected via WiFi.  

[image]

It also doesn't work via CURL

```
curl -F 'file=@firmware.bin' http://192.168.0.107/update && echo
Warning: setting file firmware.bin  failed!
curl: (26) read function returned funny value
```

jeremypoulter, 2019-12-16:
Works for me :( certainly with curl there is an error where `filename.bin` can not be read, and that is not an issue with the firmware. Maybe a permission issue?

```
$ curl -# -F file=@.pio/build/openevse_esp32-gateway-e/firmware.bin http://172.16.0.105/update | cat
################################################################################################### 100.0%
OK
```

```$ curl -# -F file=@a_file_that_is_not_found.bin http://172.16.0.105/update | cat
Warning: setting file a_file_that_is_not_found.bin  failed!
##O=#   #                                                                                                
curl: (26) read function returned funny value
```

Could be the same in the web UI, maybe check the console?

glynhudson, 2020-02-21:
It's working now 👍

commit 008cc550 (2019-09-30): Updated update GUI, should fix #25
closed 2020-02-21 by glynhudson