OpenEVSE/openevse_esp32_firmware #26: OTA upload not working

glynhudson, 2019-09-22:
I can't see to get OTA upload to work either am I missing something? I checked that `ENABLE_OTA` was enabled in the build flags. 

```
Uploading .pioenvs/openevse_huzzah32/firmware.bin
19:34:15 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '192.168.86.20', 'host_port': 33428, 'image': '.pioenvs/openevse_huzzah32/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
19:34:15 [INFO]: Starting on 0.0.0.0:33428
19:34:15 [INFO]: Upload size: 1295600
Sending invitation to 192.168.86.20 ..........
19:35:55 [ERROR]: No response from the ESP
*** [upload] Error 1
```

jeremypoulter, 2019-09-23:
This is a ESP32 SDK issue,  `esp_port': 8266` should be `3232` The latest SDK should have this fixed, but in the short term you can set `export PLATFORMIO_UPLOAD_FLAGS="-p 3232"`

glynhudson, 2019-09-28:
As a work around firmware can be updated using 

`curl -F 'file=@firmware.bin' http://<IP-ADDRESS>/update && echo`

As mentioned in https://github.com/OpenEVSE/ESP32_WiFi_V3.x/issues/20

closed 2019-09-28 by glynhudson