OpenEVSE/openevse_esp32_firmware #30: Web server HTTPS

jeremypoulter, 2019-10-01:
Enable HTTPS support for the webserver, including changing the certificate and private key.

ausil, 2020-05-29:
I use letsencrypt for certs on my emonpi, which is great. however, I am not sure that I want to expose my openevse to the internet for authentication to succeed. Perhaps there could be some tie-in to the emonpi to enable certificates to be authenticated and signed. using something like sniproxy could work well.

jeremypoulter, 2020-06-03:
@ausil Interesting, for local in-home access? or are you accessing the emonpi remotely?

ausil, 2020-06-03:
@jeremypoulter I am accessing it remotely, I have a few services I run at home behind sniproxy. I like to be able to monitor energy usage when I am traveling, I have tweaked the tls config on apache to make things as secure as I can

PR #695 (2024-08-07): Certificate management
Adding the ability to upload certificates to ESP32

Fixes #29 
Fixes #30
closed 2024-08-07 by jeremypoulter