OpenEVSE/openevse_esp32_firmware #31: Secure MQTT

jeremypoulter, 2019-10-01:
Add support for secure MQTT connections, depends on #29

glynhudson, 2020-02-04:
Load certificates for:

- mqtt.emoncms.org
- io.adafruit.com
- Amazon MQ

chris1howell, 2020-02-04:
and:

data.openevse.com

glynhudson, 2020-02-04:
@chris1howell  Do you run a MQTT server at `data.openevse.com`? This issue was specially regarding MQTTS. Posting to data.openevse.com via HTTPS is already supported. I will confirm this with @jeremypoulter

jeremypoulter, 2020-02-04:
Correct, posting to data.openevse.org via HTTPS is already supposed. MQTT is a slightly different usecase as you are more likely to be posting to an unknown service.

chris1howell, 2020-02-04:
MQTT is running at data.openevse.com. We are setting ESP32 modules to MQTT
enabled by default using HAL.getLongID() as the default topic. We are using
a default user/password for now but would like to move to a unique client
cert per device.

jeremypoulter, 2020-02-04:
I  think you are talking about client certificates right? This is not what
this covers, just connection via secure MQTT, the certs referenced are the
server root certificates.

chris1howell, 2020-02-04:
Yes, the client certificate is for authorization only replacing a static
MQTT username and password.

SSL Encryption over MQTT is needed to data.openevse.com.

jeremypoulter, 2020-02-06:
Ok let's add the client certificate support as another issue, and keep this as just connection to the server via MQTTS

jeremypoulter, 2020-02-18:
@chris1howell can you send details of how to connect with a user/pass for testing?

chris1howell, 2020-02-18:
Sure. I have dozens of 3.02 clients connected now. I will DM
username/password.

jeremypoulter, 2020-02-25:
Have this working with Amazon MQ, but I am having issues with Adafruit and data.openevse.com, reporting as not supported :( connecting with OpenSSL to both of those uses TLS1.3 which is not supported but should fall back to TLS1.2.

PR #57 (2020-03-24): Secure MQTT support
Enabled support for Secure MQTT.

Fixes #31 and #52
closed 2020-03-24 by glynhudson