OpenEVSE/openevse_esp32_firmware #33: Use NTP to set time on OpenEVSE controller

glynhudson, 2019-10-08:
As a first step towards #5 and negating the need for a hardware RTC, NTP time from the ESP could be used to automatically set the time on the OpenEVSE controller.

We should keep the option to "sync" the time from a web browser for the use case where no web connection is present.

jeremypoulter, 2020-02-18:
Two other kind of related things that should be done;

- On start up, the ESP should get the time from the EVSE, more likely to be correct than the what ever the boot up time is on the ESP
- Add endpoints for getting/setting the time on the ESP/EVSE rather than doing this directly via RAPI

glynhudson, 2020-02-21:
Implemented in https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/56

closed 2020-03-23 by glynhudson

arnolde, 2020-10-29:
@glynhudson I bought an EVSE+Wifi from your shop in Spring 2020 and in the webUI it says OpenEVSE v. 5.0.1 and Wifi firmware 3.0.2.dev. But nowhere can I find any time sync option? Neither "sync to browser" nor "use SNTP"? If I understand this issue correctly, it should be included from v3.0 right?

chris1howell, 2020-10-29:
@Ethan,

NTP was released in 3.1.0, please see the release history below and update
you WiFi firmware for the NTP features if desired.

https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases

glynhudson, 2020-10-29:
You can update the WiFi firmware to the latest version V3.3.2 via the web UI https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases

PR #56 (2020-03-23): SNTP support
Fixes #33 

To Do

- [x] Update time in UI when changed via NTP
- [x] Configure the timezone
- [x] Set ESP time from EVSE on start up
- [x] Wrap manual/browser time setting in HTTP API so ESP32 time is also set