OpenEVSE/openevse_esp32_firmware #35: Cannot compile (python 3?)

glynhudson, 2019-10-23:
I can't seem to compile the latest mater branch. I think the issue could be due to changing the build script to use python3. What are the steps I need to tell Pio to use Python3? I've got python 3.6.8 installed on my machine which in invoked by `$ python3` just `$python` invokes python 2.7.15

jeremypoulter, 2019-10-23:
I believe PIO/VSCode uses a virtual env and its own python. So you should just need to update PIO and/or the VSCode PIO extension.

glynhudson, 2019-10-23:
I'm running PlatfromIO V4.0.3 which I'm pretty sure is the latest. 

I fixed the issue be deleting the re-cloneing the repo folder. I think the issue had something to do with the GUI submodule I couldn't seem to get it to update. For future reference should I be updating the GUI sub module before trying to compile? If so how is this done? 

I retrospect this issue had nothing to do with the Python3 change.

closed 2019-10-23 by glynhudson