OpenEVSE/openevse_esp32_firmware #36: Periodically post MQTT announce message and use bare announce/ topic

glynhudson, 2019-11-01:
Building upon #15 it would be useful for Home Assistant integration for the Carbon Coop OpenDSR project that @beaylott is leading if:

 - The OpenEVSE Announce message was periodically published to the announce topic (every 30s) and 

- The announce message was published to the bare announce topic e.g `openevse/announce` rather than an ID sub topic e.g `openevse/announce/<id>`

What are your thoughts @jeremypoulter? I'm happy to make both these changes, I think it would make it easier for any service to integrate with OpenEVSE MQTT.

beaylott, 2019-11-01:
The argument in favour of the 'naked' announce topic is that it is not then necessary to sift through wildcard matched messages which maybe easier or harder depending on what is consuming the messages. In home assistant its easier to filter on the message payloads but in other platforms it might be that filtering on the sub-topic is easier.

beaylott, 2019-11-01:
On the announce message periodic posting, the other option is to post it in response to a message to some other topic. This is obviously more complex.

jeremypoulter, 2019-11-03:
On the downside you generate a lot more traffic on the MQTT broker, especially if you have multiple devices and also your discovery time is an average of half the repeat time.

I think the idea of announcing in response to a message on another MQTT topic is potentially better, still an abuse of MQTT IMHO but at least it resolves some of the negatives.

How about we keep the current method as is, I think this is the 'correct'* way to do it and as an alternative when you post say `*` to `openevse/announce/discover` any connected device can respond on `openevse/announce` with the same JSON message as send to `openevse/announce/<id>`, eg:

```
{
  "state": "connected",
  "id": "c94f330d55ac",
  "name": "openevse-55ac",
  "mqtt": "emonevse",
  "http": "http://192.168.86.20/"
}
```

Although that being said, what exactly is the issue with home assistant? Does it not support subscribing to `openevse/announce/#`? In principle this is no different than subscribing to `openevse/announce` the end result is the same, you get multiple JSON blocks for each device on that topic. I don't think you even need to look at the actual topic name.

If you really want the 4 char ID as an explicit field in the JSON that can be added also, but this can be derived from `id` (not `name` as this is user configurable).

* Please also bare in mind that using MQTT itself is not really the most appropriate way of doing device discovers, mDNS (already supported) and/or SSDP would be much better as they do not require the device to first be setup to connect to the MQTT broker.

jeremypoulter, 2019-11-03:
Actually isn't it the MQTT broker that does the topic matching for the wildcards? So I am really struggling to see what the value would be? I think you get what you want by just ignoring the topic names.

beaylott, 2019-11-04:
Ok I got confused and actually HA does support wildcards (maybe it didnt use too or something). So we dont actually need the bare announce topic.

jeremypoulter, 2019-11-04:
Ok, so I think we can close this then @glynhudson / @beaylott

beaylott, 2019-11-05:
Are you still planning to implement the periodic announce or in response to a command to openevse/announce/discover ?

glynhudson, 2019-11-05:
>Are you still planning to implement the periodic announce or in response to a command to openevse/announce/discover ?

Is this still needed if you can use wildcard to listen on the announcement sub-topic e.g `openevse/announce/#`

jeremypoulter, 2019-11-30:
I think we can close this off, correct? @glynhudson @beaylott

closed 2019-12-17 by glynhudson