OpenEVSE/openevse_esp32_firmware #37: A single base topic?

beaylott, 2019-11-01:
This is another un-important possible change...

At the moment messages are posted to 

<base topic>/

which by default is

openevse-XXXX/

but announce messages are posted to (irregardless of base topic):

openevse/announce/XXXX

(although that may change soon to openevse/announce).

For consistency does it make sense to have a separate base topic , so messages would be posted too:

openevse/<base topic>

which would then by default be:

openevse/openevse-XXXX

I guess the problem with this is that it isnt backward compatible?

Personally I dont have strong feelings about this. But I thought it would be worth raising it.

glynhudson, 2019-11-01:
Ah yes, I had noticed this. Using `openevse/openevse-XXXX` would cause issues with posted to Emoncms via MQTT since Emoncms looks for the sub-topic to create the Input names 

e.g `<Node-Name>/<Input-Name>`

jeremypoulter, 2019-11-03:
You can also configure the base topic to be `openevse/whatever` if you wish.

closed 2019-11-26 by glynhudson