OpenEVSE/openevse_esp32_firmware #38: Not plugged in notification

jeremypoulter, 2019-11-05:
A few times recently I have even noticing the car or is not properly being plugged in so in the morning the car has not been charged.

It would be really good if a notification could be sent if the car is not plugged in at some specific time, eg overnight.

Some possible options could be email, SMS, Slack or other web service.

jeremypoulter, 2019-11-18:
Possible options:

- [W3C Notifications](https://developer.mozilla.org/en-US/docs/Web/API/notification), [Google Push notifications](https://developers.google.com/web/ilt/pwa/introduction-to-push-notifications)
- Add a notification service to EmonCMS

glynhudson, 2020-06-10:
Depends on #53

jeremypoulter, 2021-11-13:
I think this is probably best implemented via an external service, eg Home Assistant, as it is a bit of a unique use case. All the required info is available via MQTT or HTTP so closing this off

closed 2021-11-13 by jeremypoulter