OpenEVSE/openevse_esp32_firmware #42: Add Support for RAPI 5.0.0

lincomatic, 2019-12-07:
I added hooks to facilitate support of OCPP. In the process, RAPI got altered a bit, and became incompatible with existing code. The changes are as follows:

1. $ST async notification replaced with $AT, which has more parameters
2. $AB async notification is now sent on boot
3. $GS has 2 new parameters added to it. Also, the first parameter, state, is now in hexadecimal, instead of decimal, to make it consistent with $AT (and the now deprecated $ST)
going forwards all async new async notifications will be of form $Ax.

I was going to create a pull request, but then realized that I don't understand your code well enough to fully test it. I've put some changes into the following commit in my own branch:

https://github.com/lincomatic/ESP8266_WiFi_v2.x/commit/da56fea96a67225d8d3bb08e3b618207b5ba97b1

Please review them, and alter as needed, at your convenience.

jeremypoulter, 2020-01-19:
@lincomatic been looking at this. Can you clarify what difference between EVSE state and Pilot state is?

jeremypoulter, 2020-01-21:
Think this is basically done baring testing, #50. I figured out the Pilot state, if I understand correctly pilot state is only those states that can be detected via the pilot signal, ie the car state not the EVSE state

lincomatic, 2020-01-21:
> Think this is basically done baring testing, #50. I figured out the Pilot state, if I understand correctly pilot state is only those states that can be detected via the pilot signal, ie the car state not the EVSE state

Basically. You can ignore pilotstate. It mainly comes into play if you have a StateTransitionReqFunc hooked in.. the pilot state is the state that we got from reading the pilot. but code can override that and choose a different EVSE state. Another example is when the vehicle is charging... it's in STATE C. But someone turns on the auth lock. the EVSE state will go to STATE B, but if the car doesn't change its state, the pilotstate will reflect that.

PR #50 (2020-01-21): RAPI 5.0.0 support
Fixes #42 and #51
closed 2020-01-21 by jeremypoulter