OpenEVSE/openevse_esp32_firmware #44: NPM Audit screaming about handlebars security problem

DrFrankReade, 2019-12-10:
When using on a Raspberry Pi Zero W  (which is a pain to get Node.js running on anyway as Pi Zero is armv6. But it's plenty fast for talking serial to the OpenEVSE over Rapi. At any rate...

Node V11.5.0
NPM V 6.4.1

Here's the report...which may entirely be my fault, so feel free to shout me down, call me an idiot and kick-ban me for life. But this is what the freshest of fresh installs yielded.

<pre>                      === npm audit security report ===                       <font color="#555753"> </font>
<font color="#555753">                                                                                </font>
# Run <span style="background-color:#FFFFFF"><font color="#000000"> npm update handlebars --depth 3 </font></span> to resolve 4 vulnerabilities
<font color="#555753">┌───────────────┬──────────────────────────────────────────────────────────────┐</font>
<font color="#555753">│</font> <font color="#FCE94F"><b>Moderate</b></font>      <font color="#555753">│</font> <b>Denial of Service</b>                                            <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Package       <font color="#555753">│</font> handlebars                                                   <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Dependency of <font color="#555753">│</font> nyc [dev]                                                    <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Path          <font color="#555753">│</font> nyc<font color="#555753"> &gt; </font>istanbul-reports<font color="#555753"> &gt; </font>handlebars                          <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> More info     <font color="#555753">│</font> https://nodesecurity.io/advisories/1300                      <font color="#555753">│</font>
<font color="#555753">└───────────────┴──────────────────────────────────────────────────────────────┘</font>

<font color="#555753">┌───────────────┬──────────────────────────────────────────────────────────────┐</font>
<font color="#555753">│</font> <font color="#EF2929"><b>High</b></font>          <font color="#555753">│</font> <b>Arbitrary Code Execution</b>                                     <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Package       <font color="#555753">│</font> handlebars                                                   <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Dependency of <font color="#555753">│</font> nyc [dev]                                                    <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Path          <font color="#555753">│</font> nyc<font color="#555753"> &gt; </font>istanbul-reports<font color="#555753"> &gt; </font>handlebars                          <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> More info     <font color="#555753">│</font> https://nodesecurity.io/advisories/1316                      <font color="#555753">│</font>
<font color="#555753">└───────────────┴──────────────────────────────────────────────────────────────┘</font>

<font color="#555753">┌───────────────┬──────────────────────────────────────────────────────────────┐</font>
<font color="#555753">│</font> <font color="#EF2929"><b>High</b></font>          <font color="#555753">│</font> <b>Arbitrary Code Execution</b>                                     <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Package       <font color="#555753">│</font> handlebars                                                   <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Dependency of <font color="#555753">│</font> nyc [dev]                                                    <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Path          <font color="#555753">│</font> nyc<font color="#555753"> &gt; </font>istanbul-reports<font color="#555753"> &gt; </font>handlebars                          <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> More info     <font color="#555753">│</font> https://nodesecurity.io/advisories/1324                      <font color="#555753">│</font>
<font color="#555753">└───────────────┴──────────────────────────────────────────────────────────────┘</font>

<font color="#555753">┌───────────────┬──────────────────────────────────────────────────────────────┐</font>
<font color="#555753">│</font> <font color="#EF2929"><b>High</b></font>          <font color="#555753">│</font> <b>Prototype Pollution</b>                                          <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Package       <font color="#555753">│</font> handlebars                                                   <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Dependency of <font color="#555753">│</font> nyc [dev]                                                    <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> Path          <font color="#555753">│</font> nyc<font color="#555753"> &gt; </font>istanbul-reports<font color="#555753"> &gt; </font>handlebars                          <font color="#555753">│</font>
<font color="#555753">├───────────────┼──────────────────────────────────────────────────────────────┤</font>
<font color="#555753">│</font> More info     <font color="#555753">│</font> https://nodesecurity.io/advisories/1325                      <font color="#555753">│</font>
<font color="#555753">└───────────────┴──────────────────────────────────────────────────────────────┘</font>
</pre>

DrFrankReade, 2019-12-10:
npm update fixes it.

closed 2019-12-10 by DrFrankReade

jeremypoulter, 2019-12-11:
Think you may be using the wrong repo, take a look at https://github.com/OpenEVSE/openevse_wifi_server, but yes, I need to do another round of NPM updates