OpenEVSE/openevse_esp32_firmware #49: Hardware info

jeremypoulter, 2020-01-05:
We should add some hardware info to the `config` so we can distinguish between the ESP8622 and ESP32

glynhudson, 2020-01-14:
Good idea, however at present V3.x is ESP32 while V2.x is ESP8266

jeremypoulter, 2020-01-14:
Correct, just would be nice to be more explicit, there is also the Node.JS version.

Niko-La, 2020-02-06:
In a similar request, it would be nice once the mqtt is established if the charger sends an initial post that could also include the local ip of the webserver.  My router does not workd with openevse.local so I have to either run fing or go to the box which is not terrible but would be a nice feature.

Maybe it cant be new rapi command like $IP?

jeremypoulter, 2020-02-07:
@Niko-La we already do this, see #15, although this info should be added to the docs.

PR #241 (2021-08-30): Expose the ESP chip info and the build env
This change exposes the ESP chip info (fixes #49) and the Platform IO build env (fixes #223) via the `/config` endpoint to help with hardware identification and selecting an appropriate software update.

Also includes a few bits of tidy-up.

Related GUI changes OpenEVSE/openevse_wifi_gui#61
closed 2021-08-30 by glynhudson