OpenEVSE/openevse_esp32_firmware #53: Detecting car plugged in while in 'Sleeping' sate

glynhudson, 2020-01-29:
Just seen this forum post which has been asked a few times before: https://community.openenergymonitor.org/t/how-to-tell-if-car-is-connected-during-timer-state-254/13111

This prompted me to think about RAPI 5.0.0 https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/50. 

I still need to test this. However, I believe the ability to detect if a car is plugged in while in 'sleeping' mode should be possible with RAPI 5. Is there an API for this? Maybe this is something we should add to the GUI? Is this doable?

jeremypoulter, 2020-01-29:
Correct, this will require RAPI 5. I have done a little test, and mostly works. I did raise lincomatic/open_evse#115 but this will only really change the speed that the connection is detected.

glynhudson, 2020-02-21:
I've been testing open_evse FW [V6.3.1](https://github.com/openenergymonitor/open_evse/releases/tag/v6.3.1) for a few days now with RAPI 5.0.0 and it seems to work fine with the latest WiFi 3.x. Nice work. 

Is it now possible to realise this issue?

glynhudson, 2020-02-26:
Here is documented the RAPI 5.0.0 response for the various `$GS` state responce including car plugged in while EVSE is sleeping:

Sleeping car not plugged in:

`$OK fe 4497 03 0200^2C`
or
`$OK fe 120 01 0200^13`

Sleeping car plugged in:
`$OK fe 4497 03 0700^29`

Ready car not plugged in:
`$OK 01 120 01 0200^11`

Ready and charging 
`OK 03 3 03 0540^12`

jeremypoulter, 2021-02-18:
This is implemented in v4.0, the MQTT and HTTP status now include `vehicle` with the state of if a vehicle is connected to the EVSE or not. Note this depends on the latest OpenEVSE firmware

glynhudson, 2021-02-19:
Fantastic! We've been shipping open_evse FW V5.xxx onwards since Aug 2020
so there will be quite a few units able to take advantage of this right
away.

Is it possible to display the state in the UI? e.g

"Waiting - EV Not connected"
"Waiting - EV Connected"

jeremypoulter, 2021-02-19:
Just remember v4 needs at least 7.1.2 of the open_evse. When completed we can maybe look at what makes sense to backport to v3 to support older open_evse firmwares

jeremypoulter, 2021-02-19:
And yeah, updating the UI is fine

closed 2021-03-07 by jeremypoulter