OpenEVSE/openevse_esp32_firmware #54: Stop charging when solar PV stops generating

glynhudson, 2020-02-11:
Currently, when charging using "Eco Mode" (solar PV divert) the EVSE will continue charging at the lowest current (6A) forever until the EV is fully charged even after the solar PV has stopped generating. 

A highly requested feature is for the charging to stop after the solar PV has stopped generating. This will allow users to then use a timer charge to topup the remaining amount required during off-peak hours. 

I propose changing default Eco Mode behaviour to start the charge once solar PV / grid export > 500W for 15min and stop the charge if solar PV / grid export < 500W for 30min. 

This feature should be backported to ESP8266. 

***

As an advance option the user could adjust these parameters, however minimum values should be set  e.g >100W for 10min stop to avoid excess contactor wear caused by regular stopping / starting. 

Another extension to EcoMode is for the charger to revert to charging at full power if the timer becomes active during an EcoMode charging session.

jeremypoulter, 2020-03-15:
Looking through the code the divert calculations are all done on current, and we have a min of 6A defined for turning on the charge (this is part of the spec right?) so would it make sense to use this for turning off (1440w), rather than 500w?

jeremypoulter, 2020-03-17:
Also see #62

glynhudson, 2020-03-18:
As discussed here is some solar PV data which can be used for testing. Data is in Watts in CSV format with datetime string:

[image]

[image]

[image]

[CSV solar PV data.zip](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/files/4348792/CSV.solar.PV.data.zip)

chris1howell, 2020-03-18:
The key is to this issue is avoiding cycling the relay on/off hundreds of
times during the course of the day.

There will likely need to be some time and power conditions that need to be
meet to justify cycling the relay in both directions.

jeremypoulter, 2020-03-18:
For openness I am going to write a wrapper around the divert module so I can replay those CSV files and see the results.

My current thinking is that we use a rolling average to smothe out the wave so hopefully won't cause a transition too often.

Will welcome any other CSVs of different usecase.

jeremypoulter, 2020-03-19:
I have been doing some initial work in Google Sheets:

https://docs.google.com/spreadsheets/d/1GQEAQ5QNvNuShEsUdcrNsFC12U3pQfcD_NetoIfDoko/edit?usp=sharing

This is just applying a simple smoothing algorithm. Has some fairly reasonable results. comments welcome.

glynhudson, 2020-03-24:
As discussed here are CSV data exports for Grid Import export feeds:

[image]

[image]

[image]

[Grid IE CSVs.zip](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/files/4375246/Grid.IE.CSVs.zip)

Zuikkis, 2020-03-24:
Perhaps there should be a way to configure EV battery size, so OpenEVSE would know how critical it is to use solar power fully?

I have 10kWp plant and I'm charging an Outlander PHEV which has takes about 9kWh from empty to full, so it's not really all that important for charge curve to follow solar output perfectly. But I don't want to use any bought electricity at day time.. 

So for me, it would be ok to have like 1-2kW safety gap to avoid going over the solar output.

jeremypoulter, 2020-03-25:
@glynhudson the Grid IE CSVs only had the solar in, not the Grid IE

jeremypoulter, 2020-03-25:
@Zuikkis The biggest problem is not necessarily knowing how much charge the car needs (although this may help) but knowing how much power you are going to get from the solar. You only know what has happened and (short of getting in to full on AI/ML) it really hard to predict what is going to happen. So putting too large buffer in there is could very well result in no charging.

glynhudson, 2020-03-26:
> @glynhudson the Grid IE CSVs only had the solar in, not the Grid IE

It's been fixed now, please download the rev2 CSV attached to the post above.

jeremypoulter, 2020-03-26:
@Zuikkis Would you be able to share a few exports of your solar system? May be good to have some examples of such a large system.

glynhudson, 2020-04-02:
As discussed, see attached solar PV + VRMS outout 

[image]

[solar-vrms.zip](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/files/4420262/solar-vrms.zip)

Zuikkis, 2020-04-02:
[Solar.zip](https://github.com/OpenEVSE/ESP32_WiFi_V3.x/files/4420355/Solar.zip)

@jeremypoulter sorry I missed you request earlier. Here are some csv's from my plant. I tried to choose different style days.

BTW, one interesting issue that maybe should be taken in consideration. Here in Finland we use three phase electricity, yet many EV's charge only with one phase. Electric companies charge each phase independently. That means that if my solar plant is producing 3000W (=1000W per phase), and I charge car with 2000W, I will actually have to pay for the 1000W... 

Currently I'm running a Raspberry Pi with Domoticz home automation, and I'm controlling the charger relay so that it starts charging when solar output is >6kW and stops when <5.5kW.. That works reasonably well with the Outlander's stock 8A (~1800W) charger.

As said before, I don't own OpenEVSE yet. But obviously it would be an improvement to have adjustable charge current.. I would be generating the MQTT output myself with the RPi, so it's not a big deal to divide the solar output by three before passing to OpenEVSE.. So 6kW would become 2kW, and then it would work as expected. But other's might have different systems where it' more difficult to intervene.

jeremypoulter, 2020-04-02:
I think in that case you would need to give OpenEVSE a feed that is just the solar generation on the same phase as the car. 

There is also a 3 phase version of EmonEVSE but that is pointless for an Outlander (I have one too)

glynhudson, 2020-04-06:
I've just tested the latest `divert_mode_off` branch. 

I'm happy to report that the charging rate successfully tracked the solar PV generation and charging successfully stopped when the solar PV dropped below 1.4kW for a period of time. 

[image]

However, the charging did not begin again when the solar PV increased above the threshold again. See the graph above which shows that the charge rate is correctly being calculated but the charging is not resumed. 

A possible explanation for this is that when the charging was stopped the EVSE entered 'disabled' state rather than 'sleeping':

[image]
[image]

jeremypoulter, 2020-04-06:
Ah, silly me will fix that.

glynhudson, 2020-04-07:
Nice work, the latest version is working well. Charge is successfully being paused and resumed 👍 

[image]

As we discussed the remaining changes require are:
- Revert current to max (previous current) when charging is paused so then charging can later be resumed by the timer at full speed
- Make solar PV (Eco Mode) always active i.e remove the auto revert to Normal Mode when EV is unplugged. Investigate displaying a message on the LCD when car is plugged in (new open_evse FW dependant) 
- UI changes: 
         - MQTT feed / numerical value for VRMS, suggest MQTT topic `emon/emonpi/vrms`
         - Ability to adjust divert parameters

glynhudson, 2020-04-09:
In RAPI 5.1.0 (firmware D6.2.1) `$GC` now returns `cmaxamps` https://github.com/lincomatic/open_evse/commit/68f362cd073827eae03f8dace6947072f428e919

`response: $OK minamps hmaxamps pilotamps cmaxamps`

*cmaxamps - max configured allowed current capacity (saved to EEPROM)*
*hmaxamps - max hardware allowed current capacity MAX_CURRENT_CAPACITY_Ln*

Maybe we should consider using `cmaxamps` (if available) instead of `hmaxamps` as the current value to restore when charging is stopped when solar PV divert is active.

jeremypoulter, 2020-05-01:
For my own reference OpenEVSE/openevse_wifi_gui#11

jeremypoulter, 2020-05-03:
@glynhudson Think this is done now, can you review #76?

Zuikkis, 2020-05-13:
Hi!

Happy new owner of OpenEVSE signing in. :) It is equipped with ESP32 wifi module, and I'm eager to try out this improved ECO mode.

This was US model, so I upgraded the base firmware to 6.2.1.EU. Wifi module is running 3.1.0-dev which seems to be quite recent version so I didn't bother flashing that yet.

Is there a pre-compiled hex of this divert_mode_off branch, or is this "soon" to be included in main branch? I do have Arduino dev environment but I haven't used it for ages, must be ancient versions. :)

glynhudson, 2020-05-13:
>Is there a pre-compiled hex of this divert_mode_off branch, or is this "soon" to be included in main branch? 

Hi, thanks for your comment. This feature is not yet merged into the master branch. There will be a pre-compiled FW release soon after it's been merged. Hopefully this should be in a week or so.

Zuikkis, 2020-05-15:
I built divert_mode_off and OTA uploaded it.

It now boots in AP mode with the default OpenEVSE ssid. I can scan for networks normally, but after that it simply reboots to AP mode again, forgetting all the settings.

I tried factory reseting the module by long pressing the button.. display showed "Factory reset" but it didn't help anything.

I'll try flashing normal firmware again. :)

Zuikkis, 2020-05-16:
I now connected ESP32 with cable for easier programming.

I can build "master" branch and it works fine. I even modified it to publish button state with MQTT, and it is working.

If I build "divert_mode_off" branch, it seems to work but only in AP mode. There is no way to make it connect to my home wifi. All other features are working.

Well, obviously this will be sorted out once you merge the branches, but I'm an impatient person. :)

jeremypoulter, 2020-05-16:
That branch has had a rewrite of the config so may have issues. just finishing merging with the master branch and I will take a look

jeremypoulter, 2020-05-17:
@Zuikkis There was a bug in saving the WiFi details, fixed now

Zuikkis, 2020-05-18:
Works brilliantly now, thanks a lot! :D

Only minor issue, I can't change the hostname. It always reverts to openevse-nnnn. Perhaps some config changes there as well?

jeremypoulter, 2020-05-18:
Thanks for reporting, will take a look at it

jeremypoulter, 2020-05-19:
Fixed the migration issue, config should be preserved now, and setting the host name

jeremypoulter, 2020-06-09:
Changes merged, #76

closed 2020-06-09 by jeremypoulter