OpenEVSE/openevse_esp32_firmware #58: Flash of 3.0.3 via USB causes non-bootable ESP32

linuxkidd, 2020-02-29:
New EVSE received with ESP32 Huzzah running 3.0.2dev version.  RAPI commands were non-functional over MQTT.  Since we need RAPI over MQTT for our project, we flashed 3.0.3 over USB TTL.

Flash 3.0.3 via USB TTL resulted in non-bootable ESP32, monitoring via serial shows the following repeated:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

Attempted to flash with all available releases ( 3.0.2b, 3.0.1b and 3.0.0b ), all with same results.

Note that I also performed a verify with esptool and it confirms the firmware hash matches.

linuxkidd, 2020-02-29:
Originally flashed using specified command line:
`esptool.py --port /dev/ttyUSB2 write_flash 0x000000 firmware.bin`

After realizing that the boot read error was looking for address 0x1000, attempted to flash with:
`esptool --port /dev/ttyUSB2 write_flash 0x1000 firmware.bin`

Now, we have a different boot error:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:496576
ets Jun  8 2016 00:22:57
```

linuxkidd, 2020-02-29:
Ok, seems using the documented flash procedure erased the bootloader.  I recovered by doing a debug level flash of the ESP32 with an example program in Arduino IDE, then flashing 3.0.3 via esptool to offset 0x10000.

Recovered with the following command to re-write the bootloader and the correct 3.0.3 firmware:
`esptool.py --port /dev/ttyUSB2 --before default_reset --after hard_reset write_flash -z --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader_dio_40m.bin 0x10000 firmware-3.0.3.bin`

The boot_app0.bin, bootloader_dio_40m.bin files are from arduino ide with ESP32 board support installed.

Hope this helps someone else!

jeremypoulter, 2020-02-29:
Going to reopen this, looks like we need to document the process, I mostly use PlatformIO for flashing or the OTA update in the UI but the docs to flash via esptool should be correct

linuxkidd, 2020-02-29:
@jeremypoulter I completely agree.  I believe the proper syntax when the ESP32 is already functional (aka, the bootloader wasn't wiped) is as follows:
`esptool.py --port $PORT write_flash 0x10000 firmware.bin`

I ordered a batch of Huzzah 32's which should be in next week.  I'll test and report back unless this is resolved before then.

linuxkidd, 2020-02-29:
Updated issue description to more accurately reflect nature of the issue...

chris1howell, 2020-03-02:
This would explain why PyFlasher is no longer working. It defaults to 0x00000. Is there a reason for the nonstandard 0x10000 offset?

jeremypoulter, 2020-03-02:
Nothing specific to this project, is the default for the Arduino on ESP32

commit 8a74e174 (2020-03-10): Correct ES32 esptool upload instructions fix #58
closed 2020-03-10 by glynhudson