OpenEVSE/openevse_esp32_firmware #59: Feature Request: Loss of MQTT connection disables EVSE

linuxkidd, 2020-02-29:
I have an open source project for RV'ers which permits load sharing between their RV and EV charger.  ( EV4RV.org )  The system works by adding an energy monitor to the RV's incoming power feed and calculating 'spare' current, after deducting 10% overhead for the breaker, it sends the remainder value over MQTT RAPI  ( $SC ) to the OpenEVSE.

The code detects an absence of current feed from the energy monitor and disables the EVSE ( $FD ) on this loss of feed and re-enables the EVSE ( $FE ) when energy readings return.  However, there's nothing that protects the upstream breaker should the OpenEVSE lose comms to the MQTT server.

The request is -- An option provided in the MQTT config dialog to disable the EVSE after X seconds of not MQTT connection ( 30 seconds seems sane ), then re-enable once a connection is re-established.

jeremypoulter, 2020-02-29:
I was actually thinking about a similar use case for when we move the timer
logic on to the ESP.

What I was thinking in that case was maybe the ESP (or in your case the
system managing the charge) could set a time limit of say charge for 30mins
every 15mins during the period of the charge. That way it would operate as
you suggested should connection be lost.

I have not tried this but maybe worth a try.

Jeremy

linuxkidd, 2020-03-01:
Hi Jeremy,
  Unfortunately, 15 and 30 minutes is not super comfortable for us.  RV loads can vary wildly even inside a minute's time ( Any one of 3 roof mount AC kick's on -- +13 amps, Load of laundry moves from washer to dryer, another +13 amps ).  With the WiFi based energy monitors we're using, we get current values every 1 second and can adjust the charge rate at that interval.

Michael

jeremypoulter, 2020-03-01:
The power and timer are independent of each other you can update the (Max) power as frequently as you like (but be aware it will take some time for the car to update the power usage). 

The timer is just used to provide the safety feature you are looking for, and you can adjust the times to whatever suits your needs. In my example you are essentially saying charge for another 30 mins every 15 mins, but could equally be charge for 2mins every 1 min. As the timer is undated before the timer finishes the timer never actually stops and the charger will behave as if there was no timer set (in theory).

closed 2026-09-21 by github-actions[bot]