OpenEVSE/openevse_esp32_firmware #61: Potentail webserver stability issues

glynhudson, 2020-03-16:
I've been unable to replicate myself but a number of users have reported connectivity issues using the ESP32 with the latest dev FW. One user reported being able to successfully control the EVSE via demand shaper (MQTT) while being unable to access the web UI. This indicates a possible issue only effecting the webserver component. 

https://community.openenergymonitor.org/t/esp-32-beta-loss-of-wifi-connection/13478/14

chris1howell, 2020-03-16:
There are currently 21 stations with version 3.0.2 connected to the OpenEVSE MQTT server. They all seem to disconnect after a couple days and require a reboot to restore the Web interface and MQTT.

Not sure it is related but while working with Sam on the Tesla client he experienced a lot of WDT task crashes. The client would crash after a few minutes. The fix was a vTaskDelay(10) which now allows the tesla client to run a couple days.

  if (_respPhase == 1) {
      while (_client.available()) {
	vTaskDelay(10);
	_resp.respStr += _client.readString();
      }

jeremypoulter, 2020-03-16:
Where is that code?

chris1howell, 2020-03-17:
The code is here in the Tesla Client.

https://github.com/lincomatic/ESP32_WiFi_V3.x/blob/master/src/tesla_client.cpp

jeremypoulter, 2020-03-18:
Ah, you have just highlighted that main reason why we use Mongoose and not the WiFiClientSecure ;-)

jeremypoulter, 2020-03-18:
@MarkNinja I noticed this commit on your fork https://github.com/MarkNinja/ESP32_WiFi_V3.x/commit/ff9cc88176d59d968ccfdeddd73499601ec88cd2. Does this resolve similar issues to the ones highlighted here?

MarkNinja, 2020-03-18:
Yes.  I believe my WiFi router was dropping the connection occasionally and the ESP32 would occasionally not be reachable.  Since my change to retry the WiFi connection when dropped I have not had the issue.

jeremypoulter, 2020-03-26:
Hopefully this is closed now

closed 2020-03-26 by jeremypoulter