OpenEVSE/openevse_esp32_firmware #68: ESP32_WiFi Take over LCD

chris1howell, 2020-04-20:
The ESP32 module now has much more rich information than the OpenEVSE controller.

During Boot the ESP32 takes over writing to the LCD for several seconds to communicate WiFi SSID, Hostname and IP address. I propose keeping the ESP32 in control of LCD. The top line should stay the same with Status Service Level,Pilot current. The bottom line should cycle through important information relevant to the current state such as:
Power added 2.1kw
Internal Temp 48C
Elapsed time 1:23
Battery SOC 73%  (if available)
Target SOC 85% (if available)
Time left 1:15 (if available)
Range 208 (if available)
Range Added 58 (if available)

jeremypoulter, 2020-04-20:
Think this should be possible, I think this will need to be dependant on RAPI 5.0.0 to have the greater visibility of the menu state. @glynhudson and I have also talked about something similar for #54.

@glynhudson and I have also discussed connecting the LCD to the ESP and doing the full menu so that code can be removed from the OpenEVSE firmware (should save a load of code space)

chris1howell, 2020-08-25:
State = 00  WHITE 

State = 01  GREEN
Line 0     "Ready     L2:48A" 
Lines 1    "Energy   1,018Wh"
           "Lifetime 2313kWh"
           "EVSE Temp  30.5C"
           "Time     03:14PM"
           "Date  08/25/2020"

State = 02  YELLOW
Line 0     "Connected L2:48A" 
Lines 1    "Energy   1,018Wh"
           "Lifetime 2313kWh"
           "EVSE Temp  35.5C"           
           "Time     03:14PM"
           "Date  08/25/2020"

State = 03  BLUE
If TEMP > 60 YELLOW
Line 0     "Charging   47.8A" 
Lines 1    "Energy   1,018Wh"
           "Lifetime 2313kWh"
           "EVSE Temp  45.5C"
if Tesla   "Range  259 miles"
           "Charge level 79%"
           "Charge limit 85%"
           "Added  127 miles"
           "Voltage  243v AC"

State = FE PURPLE
Line 0     "zzZ Sleeping Zzz" 
Lines 1    "Time     03:14PM"
           "Date  08/25/2020"
if timer   "Start    10:00PM"
           "Stop     06:00AM"

Error States RED

State = 04 
Line 0     "VEHICLE ERROR   "
Line 1     "VENT REQUIRED   " 

State = 05 
Line 0     "VEHICLE ERROR   "
Line 1     "VEHICLE CHECK   " 

State = 06 
Line 0     "SAFETY ERROR    "
Line 1     "GROUND FAULT    " 

State = 07 
Line 0     "SAFETY ERROR    "
Line 1     "GROUND MISSING  " 

State = 08 
Line 0     "SAFETY ERROR    "
Line 1     "STUCK RELAY     " 

State = 09 
Line 0     "SAFETY ERROR    "
Line 1     "SELF TEST FAILED" 

State = 0A 
Line 0     "SAFETY ERROR    "
Line 1     "OVER TEMPERATURE" 

State = 0B 
Line 0     "SAFETY ERROR    "
Line 1     "OVER CURRENT    " 

State = FF 
Line 0     "DISABLED        "

glynhudson, 2020-09-02:
If solar PV divert enabled:

Line0 "Eco Mode Enabled"
Line 1 "PV:xx.xkW EV:xxA"

chris1howell, 2020-09-02:
To keep the interface consistent, we should not mess with Line 0 which
displays important information State Service Level and Current.

Divert is not just for PV, it can being used for any generation source,
surplus or excess capacity. We really should move away from calling it
Solar, beacuse it can be used in so many ways.

I would suggest using line 1 only and use the same units and changing the
message every xx seconds. Something like this.

Line 1 "Divert  xxxx.x Kw"
Line 1 "Output xxxx.x Kw"

closed 2021-03-07 by jeremypoulter