OpenEVSE/openevse_esp32_firmware #71: Add Vehicle UI Tab

glynhudson, 2020-04-21:
After https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/67 is merged we should make a 'Vehicle' tab on the web UI interface to interface with the Tesla integration e.g set enter username / password etc. I've added the API call for the Tesla integration as a comment in #67 

The vehicle tab should be structured to allow for multiple vehicle integrations, probably with only a single integration being used at a time..for the moment.  

OVMS should be the next integration to consider adding.

Pukkahq, 2020-04-26:
I have not tested this in the last few years but this might help for the BMW
https://github.com/openenergymonitor/emonhub/tree/emon-pi/conf/interfacer_examples/bmw

Pukkahq, 2020-04-26:
just been and added the interface yes it still works
[image]

glynhudson, 2021-07-22:
The GUI looks great, however where is the MQTT topic for SoC set? Or is this pre-set to a particular topic? If this is the case the GUI should have an info box to let the user know what topic to use for SoC 

[image]

jeremypoulter, 2021-07-22:
Not quite sure I understand the issue, your screen shot has the input field for the SoC topic, is it not clear that you should enter a topic name?

glynhudson, 2021-07-22:
Apologies, my mistake, I didn't realise the fields where for topic names. I figured it out, working now :-) 

To make it obvious I've added the word  'topic' as a suffix to the field labels: https://github.com/OpenEVSE/openevse_wifi_gui/commit/a50ad7706d1e2a69c532c3f3704ff16a5e304fe9

[image]

[image]

PR #198 (2021-07-20): Adding a GUI for configuring the vehicle data source
Fixes #71

Addin support for getting configuring the vehicle data source. Getting data from the Tesla API and MQTT are currently supported. 

Displayed on the UI in the new vehicle tab as well as on the LCD.
closed 2021-07-22 by glynhudson