OpenEVSE/openevse_esp32_firmware #72: NTP Server

Pukkahq, 2020-04-30:
On system page NTP Server does not save after reboot

[image]

mikeh2010, 2020-04-30:
On main page - Not able to use Time or Energy select boxes 
[image]

glynhudson, 2020-04-30:
Could you post the output of `http://<openevse- ip-address>/status`? This will show if the custom NTP server settings have actually been stored. What custom NTP server are you trying to use? The default is `pool.ntp.org`

Pukkahq, 2020-04-30:
{"mode":"STA","wifi_client_connected":1,"eth_connected":0,"net_connected":1,"srssi":-90,"ipaddress":"192.168.1.216","emoncms_connected":1,"packets_sent":417,"packets_success":417,"mqtt_connected":1,"ohm_hour":"NotConnected","free_heap":246728,"comm_sent":6309,"comm_success":6309,"rapi_connected":1,"amp":0,"pilot":32,"temp1":225,"temp2":-2560,"temp3":-2560,"state":254,"elapsed":327,"wattsec":1616534,"watthour":0,"gfcicount":0,"nogndcount":0,"stuckcount":0,"divertmode":1,"solar":0,"grid_ie":43,"charge_rate":0,"divert_update":4,"ota_update":0,"time":"2020-04-30T13:59:44Z","offset":"+0000"}

glynhudson, 2020-04-30:
Apologies, I meant `/config`. Please could you post the output

Pukkahq, 2020-04-30:
"sntp_host":"pool.ntp.org" is shown in the config. if i change it to uk.pool.ntp.org and reboot it still shows pool.ntp.org

glynhudson, 2020-04-30:
Ok, this is a bug. Thanks for investigating. We will get this fixed.

jeremypoulter, 2020-05-03:
Fixed as part of #76

jeremypoulter, 2020-06-09:
This has now been merged

closed 2020-06-09 by jeremypoulter