OpenEVSE/openevse_esp32_firmware #74: 3.1.0 build error

mikeh2010, 2020-04-30:
Looks like the git submodule is not there so no gui...

PS D:\OPENENERGYMONITOR\ESP32_WiFi_V3.x-3.1.0> git submodule update --init
fatal: not a git repository (or any of the parent directories): .git
PS D:\OPENENERGYMONITOR\ESP32_WiFi_V3.x-3.1.0>

chris1howell, 2020-04-30:
It is my understanding the GUI only needs to be rebuilt if you wish to make
modifications to the GUI HTML, JavaScript or CSS source code. If you do not
there is no need to rebuild.

If you do there is instructions in readme for building the GUI.

mikeh2010, 2020-04-30:
Indeed...

```
If the gui directory is empty, use the following to retrieve the GUI source and fetch the dependencies:

git submodule update --init
cd gui
npm install
```

jeremypoulter, 2020-04-30:
Did you download an archive of the source from the releases page or `git clone` the repo? The former strips all the Git meta data so git commands no longer work.

mikeh2010, 2020-04-30:
Yes, I used the 'git clone' to get the repo...

git clone https://github.com/OpenEVSE/ESP32_WiFi_V3.x

jeremypoulter, 2020-05-02:
Does any other git command work? `git pull` for example?

mikeh2010, 2020-05-03:
git pull appears to be working...

```
PS D:\OPENENERGYMONITOR\ESP32_WiFi_V3.x> git pull
Already up to date.
PS D:\OPENENERGYMONITOR\ESP32_WiFi_V3.x>
```

```
PS D:\OPENENERGYMONITOR\ESP32_WiFi_V3.x> git submodule update --init
Submodule 'gui' (https://github.com/openevse/openevse_wifi_gui) registered for path 'gui'
Cloning into 'D:/OPENENERGYMONITOR/ESP32_WiFi_V3.x/gui'...
```

jeremypoulter, 2020-05-03:
> It worked this time!
> Did you do something?

Hum, odd to did not do anything. not quite sure what went wrong originally if it was properly checked out from Git.

> npm build not working!

Do what NPM says ;-) `npm run build`

mikeh2010, 2020-05-04:
Yes, I did that ;-) Just pointing out the build instructions are now wrong!

closed 2020-05-04 by mikeh2010

jeremypoulter, 2020-05-11:
Which build instructions are you looking at?

mikeh2010, 2020-05-11:
Hum, cant find it now. Maybe I got it wrong. I usually cut 'n paste though...

jeremypoulter, 2020-05-11:
I type `npm build` all the time so distinctly possible I have the wrong instructions somewhere