OpenEVSE/openevse_esp32_firmware #75: Eco to Normal mode

Pukkahq, 2020-05-02:
No sure how to diagnose this. 

When in eco mode after a period of time (unknown at this time) charging goes back to normal mode

jeremypoulter, 2020-05-03:
In the current `master` eco mode is turned off when the charging finishes. we currently have updates to the divert code that will enable 'eco' mode to be left on permanently, see the #54 / `divert_mode_off` branch

Pukkahq, 2020-05-04:
The charging did not finish, the OpenEVSE webpage had gone back to normal mode. I have battery storage and this is filled before the car starts to charge, the blue circle is when i noticed it click a few buttons and restarted the ESP32.
I need some more sunny days in the UK to test further
[image]

Pukkahq, 2020-05-04:
here is an overlay of our solar production
[image]

Pukkahq, 2020-06-01:
Here is another screenshot, I am not sure what is going on but it is turning off eco mode

[image]

jeremypoulter, 2020-06-06:
Is this with the master branch? If so there is an imminent update coming should improve matters.

Pukkahq, 2020-06-06:
yes i am on the master branch. I would test one of the other branches but struggling with platformio

jeremypoulter, 2020-06-10:
@Pukkahq we have merged the divert updates to master now, can you give them a go and let us know how it goes

glynhudson, 2020-06-10:
@Pukkahq you can use the pre-compiled V3.2.0 beta release and update via the web interface to easily test the new version. See changelog and download release: https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases/tag/3.2.0

Pukkahq, 2020-06-13:
I will report back soon, just installed it. need an empty car and sunny day now :)

glynhudson, 2020-06-18:
3.2.0 issues being discussed in #90

closed 2020-06-18 by glynhudson