OpenEVSE/openevse_esp32_firmware #77: Make ESP32 gpio available to MQTT

Zuikkis, 2020-05-14:
Hi,

small "feature request".. I have a Raspberry Pi running Domoticz home automation. I'd like to have a few buttons on the OpenEVSE box, which would just signal Domoticz to do stuff.

There's plenty of free GPIO pins on the esp32. If I wire buttons to those, is there any way to read them remotely? Ideal would be if any change of state would send MQTT message.

glynhudson, 2020-05-14:
This is possible, you would just need to add the code to read from the button and post the MQTT. There are plenty of esp code examples for how to do this. Since this is a niche use case I'm afraid this is not a priority feature for us to add. But feel free to customise your own unit.

closed 2020-05-14 by glynhudson

Zuikkis, 2020-05-14:
Oh! I don't think I bother. I'll just install some Sonoff Tasmota box next to OpenEVSE and read the button from there..

What I really wanted though, some super easy way to set different charging modes:

1) Need car later today: NORMAL mode with full power
2) Need car tomorrow: ECO mode + timer enabled for next night
3) Need car several days later: ECO mode + timer disabled

I kinda thought the included OpenEVSE button would do something like this, but it seems to be for something else. :)

glynhudson, 2020-05-15:
The charging mode can be set via MQTT,  `{base-topic}/divertmode/set` (1 = normal, 2 = eco).

We have an update for Eco+ mode which will pause the charging when solar PV stops generating below a certian level coming soon, see https://github.com/OpenEVSE/ESP32_WiFi_V3.x/issues/54

Zuikkis, 2020-05-15:
Yeah, the updated Eco+ mode was the reason I bought OpenEVSE. :) 

Setting the charging mode through MQTT is easy, but  I'd like to be able to do it easily using the button on the EVSE itself.

Zuikkis, 2020-05-16:
Turns out this was far easier to add than I thought. One line added to src/net_manager.cpp, near line 454:

```
  if(wifiButtonState != button)
  {
    wifiButtonState = button;
    if(WIFI_BUTTON_PRESSED_STATE == button) {
      DBUGF("Button pressed");
      wifiButtonTimeOut = millis();
      apMessage = false;
    } else {
      DBUGF("Button released");
      if(millis() > wifiButtonTimeOut + WIFI_BUTTON_AP_TIMEOUT) {
        startAP();
      } else {
        if (config_mqtt_enabled()) mqtt_publish(F("button:1"));  // ** Added this line
//        display_state();                                       // **Removed this line
      }
    }
  }

```

Now when I press the button on wifi module, I correctly get "openevse/button/1" mqtt message. :)

I commented out the "display_state()" as I plan to send out lcd RAPI commands from Domoticz, when pressing the button. So they would interfere probably..

Other functionality of the button is still preserved.