OpenEVSE/openevse_esp32_firmware #78: building fails

ausil, 2020-05-16:
I am probably doing something wrong. but trying to build current git head fails with 
`*** [src/web_static/web_server.lib.js.h] UnicodeDecodeError : 'ascii' codec can't decode byte 0xef in position 13009: ordinal not in range(128)
Compiling .pio/build/openevse_huzzah32_idf/bootloader/bootloader_support/src/esp_image_format.c.o
Traceback (most recent call last):
  File "/home/dennis/.platformio/packages/tool-scons/script/../engine/SCons/Action.py", line 1054, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/home/dennis/code/ESP32_WiFi_V3.x/scripts/extra_script.py", line 59, in data_to_header
    output += text_to_header(file)
  File "/home/dennis/code/ESP32_WiFi_V3.x/scripts/extra_script.py", line 30, in text_to_header
    output += u"\n  \"{}\\n\"".format(line.replace('\\', '\\\\').replace('"', '\\"'))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 13009: ordinal not in range(128)`

I have a fresh install of atom

jeremypoulter, 2020-05-17:
What version of Python do you have installed?

ausil, 2020-05-17:
@jeremypoulter I have 3.8. I am running on Fedora 32

ausil, 2020-05-18:
Sorted it out, I had python-2.7 installed for gimp also, and atom used it for its venv. removing that and restarting gave me python3 and not python2

closed 2020-05-18 by ausil