OpenEVSE/openevse_esp32_firmware #79: MQTT openevse/wh does not update realtime

Zuikkis, 2020-05-17:
Hi,

First of all there's a small bug in Dev documentation:

openevse/wh                     Calculated watthours for the current session

In reality "openevse/wh" is the total watthours, not session. I assume this is the intended behaviour, so this is just documentation error?

But also, "openevse/wh" does not increment when charging. It only updates after current session has been finished. So it's not actually "total watthours" because it's missing the watthours used in current session!

This looks funny on my Domoticz system because it calculates "daily kwh" based on this MQTT data, and now it's showing "0kWh used today" even though the car has been charging for three hours already..

I could calculate "Total+session kwh" in Domoticz, but session watthours are not available through MQTT, only with HTTP or RAPI calls. Perhaps you could add "openevse/wattsec" for session wattsecs, or modify "openevse/wh" to include session watthours?

glynhudson, 2020-05-23:
>In reality "openevse/wh" is the total watthours, not session. I assume this is the intended behaviour, so this is just documentation error?

Thanks, yes this is intended behaviour. Where in the documentation did you read this? 

> So it's not actually "total watthours" because it's missing the watthours used in current session!

Indeed

Alternatively you could post the 'amps' to Emoncms then using emocms input processing convert to power by multiplying by a VRMS feed then convert to kwh using power_to_kwh input process. You could then post the result to MQTT for use by Domoticz.

Zuikkis, 2020-05-23:
> Thanks, yes this is intended behaviour. Where in the documentation did you read this?

https://github.com/OpenEVSE/ESP32_WiFi_V3.x/blob/master/Developers_Guides/Developers%20Guide_MQTT.pdf

> Alternatively you could post the 'amps' to Emoncms then using emocms input processing convert to power by multiplying by a VRMS feed then convert to kwh using power_to_kwh input process. You could then post the result to MQTT for use by Domoticz.

I think Domoticz can do this already, I just noticed that there's a configuration option:
Energy read: Computed/From device

Now I have it set to "From device", but I'll experiment with "Computed". Car is now fully charged so can't test. :) "From device" would be better in that it then matches the OpenEVSE value, "Computed" can have differences. 

Domoticz already shows "current power" which is taken from "amps" feed.  This is realtime value which updates every 30 seconds (when mqtt message is received). But kwh value is different and updates only every hour.

This is more a problem now with "divert_mode_off" and ECO mode. For example now I haven't used my car for five days, but it's still plugged in. In reality it was already fully charged at Tuesday, but the watthours still are not added to total. When I unplug, Domoticz will show the full charge kwh added at single hour today.

I see that also the kwh value shown in OpenEVSE LCD display also behaves like this, only updates when charging is finished. So if this is a bug at all, it's already in OpenEVSE firmware and not here in esp32, so this issue is in wrong place. Sorry. :) I'll close this..

closed 2020-05-23 by Zuikkis