OpenEVSE/openevse_esp32_firmware #80: Tesla Integration Issues

glynhudson, 2020-05-18:
After testing https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/67 for the past week or so I've noticed a view issues: 

1. JSON error on the `/status` output and some related Tesla API calls:

[image]
[image]
[image]

```
{"mode":"STA","wifi_client_connected":1,"eth_connected":0,"net_connected":1,"srssi":-71,"ipaddress":"192.168.86.43","emoncms_connected":1,"packets_sent":20131,"packets_success":20115,"mqtt_connected":1,"ohm_hour":"NotConnected","free_heap":243412,"comm_sent":302554,"comm_success":302554,"rapi_connected":1,"amp":0,"pilot":32,"temp1":280,"temp2":-2560,"temp3":-2560,"state":254,"elapsed":3473,"wattsec":22493407,"watthour":51536,"gfcicount":0,"nogndcount":0,"stuckcount":0,"divertmode":1,"solar":0,"grid_ie":0,"charge_rate":7,"divert_update":2,"ota_update":0,"time":"2020-05-11T23:45:42Z","offset":"+0000""batteryRange":185.38,"chargeEnergyAdded":21.31,"chargeMilesAddedRated":86.00,"batteryLevel":77,"chargeLimitSOC":90,"timeToFullCharge":0,"chargerVoltage":0,}
```

2. No enabled / disabled status in `/config`, this will become an issue when we come to build the UI  for this feature?

3. Laggy UI, I've noticed when this feature is enabled the UI and RAPI calls can be unresponsive, requiring the interface to be reloaded to correctly display the EVSE status. More testing needed to figure out of this feature is the case of this. 

4. Token expiry? After about a week the Tesla info disappeared from `/status/ and `/config` displays `tesla_user: "",`, my Tesla login seems to have been forgotten. The serial debug periodically displays `requestAccessToken()`. Is it possible for the Tesla login expire? 

5. Cannot upload via Arduino OTA upload since the Tesla API call loop interrupts the upload. The HTTP upload via the interface still works ok. 

6. Often seeing `404 Not Found` errors with full server response via the serial debut  

```
getChargeState() vehidx=0
18-May-20, 19:12:00: Free memory 202052 - diff 42892 50584
resp
404 Not Found
_HEADER[Server]: nginx
_HEADER[Date]: Mon, 18 May 2020 18:12:01 GMT
_HEADER[Content-Type]: application/json; charset=utf-8
_HEADER[Content-Length]: 60
_HEADER[Connection]: keep-alive
_HEADER[X-Frame-Options]: SAMEORIGIN
_HEADER[X-XSS-Protection]: 1; mode=block
_HEADER[X-Content-Type-Options]: nosniff
_HEADER[Cache-Control]: no-cache
_HEADER[X-Request-Id]: aa3a486c28a9608e1907d4be2fb95526
_HEADER[X-Runtime]: 0.031904
{"response":null,"error":"not_found","error_description":""}
```
and if the car is sleeping I get the following error (accompanied by the full nginx response), waking the car up via the app and the errors goes away
```
408 Request Timeout
{"response":null,"error":"vehicle unavailable: {:error=>\"vehicle unavailable:\"}","error_description":""}
```

Have you @chris1howell  or @lincomatic  experienced any of these issues?

jeremypoulter, 2020-05-18:
I think I may have fixed the config related issues it the merge to the divert branch

glynhudson, 2020-06-26:
Most of these issues have now been fixed, will open separate issue tickets from remaining issues

closed 2020-07-16 by glynhudson