OpenEVSE/openevse_esp32_firmware #85: 3.2 MQTT hang

McGreggor, 2020-06-12:
Sorry if this isn't right place/time for pre-release issues, but I've found that with 3.2 build (on wired ethernet ESP32), MQTT/Solar PV divert can freeze on last Grid (+I/-E) value when you disable 'reject self-signed certificates) setting.  In my case, I'm only using mqtt: rather than mqtts: but this option can still be enabled/disabled.

jeremypoulter, 2020-06-12:
hum, that is strange... can you post the content of `http://<openevse name/ip>/config` ?

McGreggor, 2020-06-13:
`{"firmware":"5.1.0.T2","protocol":"4.0.1","espflash":4194304,"version":"3.2.0","diodet":0,"gfcit":0,"groundt":0,"relayt":0,"ventt":0,"tempt":0,"service":2,"scale":220,"offset":0,"mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http","https"],"ssid":"","pass":"","www_username":"","www_password":"","hostname":"evse1.fqdn","sntp_hostname":"ntp1.fqdn","time_zone":"Europe/London|GMT0BST,M3.5.0/1,M10.5.0","emoncms_server":"https://data.openevse.com/emoncms","emoncms_node":"","emoncms_apikey":"","emoncms_fingerprint":"","mqtt_server":"10.0.0.89","mqtt_port":1883,"mqtt_topic":"openevse/evse1","mqtt_user":"","mqtt_pass":"","mqtt_solar":"","mqtt_grid_ie":"fronius/p_grid","mqtt_vrms":"/fronius/voltage","mqtt_announce_topic":"openevse/announce/d1f4","ohm":"","divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_username":"","tesla_password":"","tesla_vehidx":-1,"flags":1546,"emoncms_enabled":false,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"mqtt_protocol":"mqtt","charge_mode":"eco"}`

glynhudson, 2020-06-18:
I've been unable to replicate this issue. See screenshots with Grid IE and Reject self-signed certificates enable. MQTT status is still regularly updating, see MQTT explorer screenshot:

[image]

[image]

closed 2020-06-26 by glynhudson