OpenEVSE/openevse_esp32_firmware #86: 3.2 /savetesla missing?

linuxkidd, 2020-06-13:
Attempted to enable the Tesla API, but /savetesla returns 'Not Found' accompanied by HTTP 404, no matter what options are presented.

curl test:
```
# curl -v 'http://openevse/savetesla?enable=true'
*   Trying <ip removed>:80...
* TCP_NODELAY set
* Connected to openevse (<ip removed>) port 80 (#0)
> GET /savetesla?enable=true HTTP/1.1
> Host: openevse
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 404 Not Found
< Server: Mongoose/6.14
< Connection: close
< Content-Type: text/plain
< Content-Length: 9
< 
* Closing connection 0
```

/status output:
```
{"mode":"STA","wifi_client_connected":1,"eth_connected":0,"net_connected":1,"srssi":-58,"ipaddress":"removed","emoncms_connected":true,"packets_sent":45,"packets_success":45,"mqtt_connected":1,"ohm_hour":"NotConnected","free_heap":235252,"comm_sent":1137,"comm_success":1080,"rapi_connected":1,"amp":0,"voltage":249.28,"pilot":23,"temp1":315,"temp2":false,"temp3":false,"state":2,"elapsed":1131,"wattsec":7985672,"watthour":754371,"gfcicount":0,"nogndcount":1,"stuckcount":0,"divertmode":1,"solar":0,"grid_ie":0,"charge_rate":0,"divert_update":1367,"ota_update":0,"time":"2020-06-13T03:30:48Z","offset":"-0700"}
```

/config output:
```
{"firmware":"5.0.1","protocol":"4.0.1","espflash":4194304,"version":"3.2.0","diodet":0,"gfcit":0,"groundt":0,"relayt":0,"ventt":0,"tempt":0,"service":2,"scale":220,"offset":0,"mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http","https"],"ssid":"removed","pass":"_DUMMY_PASSWORD","www_username":"","www_password":"","hostname":"openevse","sntp_hostname":"pool.ntp.org","time_zone":"America/Denver|MST7MDT,M3.2.0,M11.1.0","emoncms_server":"removed","emoncms_node":"removed","emoncms_apikey":"_DUMMY_PASSWORD","emoncms_fingerprint":"","mqtt_server":"removed","mqtt_port":1883,"mqtt_topic":"openevse","mqtt_user":"","mqtt_pass":"","mqtt_solar":"","mqtt_grid_ie":"","mqtt_vrms":"openevse/volts","mqtt_announce_topic":"openevse/announce/229c","ohm":"","divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_username":"","tesla_password":"","tesla_vehidx":-1,"flags":11,"emoncms_enabled":true,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":false,"mqtt_protocol":"mqtt","charge_mode":"fast"}
```

linuxkidd, 2020-06-13:
Confirmed in 3.2.0 source tarball and tag 3.2.0:
https://github.com/OpenEVSE/ESP32_WiFi_V3.x/blob/3.2.0/src/web_server.cpp#L1091-L1126

jeremypoulter, 2020-06-15:
The `/savexxxx` APIs have been replaced by a generic `POST` to the `/config` endpoint. I did not think anyone was using the `/savetesla` endpoint yet so was removed, sorry about that. Are you able to switch over to the `/config` endpoint?

linuxkidd, 2020-06-16:
Ah, no worries... consider this then a doc bug since the docs still show /savetelsa as the proper means of setting up the Tesla integration.

glynhudson, 2020-06-18:
> The `/savexxxx` APIs have been replaced by a generic `POST` to the `/config` endpoint. I did not think anyone was using the `/savetesla` endpoint yet so was removed, sorry about that. Are you able to switch over to the `/config` endpoint?

The `/savetesla` API was used in the beta Tesla integration as documented here: https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/67#issuecomment-617315756

e.g `http://openevse-55ad/savetesla?user=email@email.com&pass=xxxxxxxxx&enable=true`

@jeremypoulter what is the new updated API for saving a Tesla username/password? I will update the docs and create proper documentation for Tesla integration in the readme.

jeremypoulter, 2020-06-19:
So now you have to `POST` to the `/config` endpoint:

```
###

POST {{baseUrl}}/config HTTP/1.1
Content-Type: application/json

{
  "tesla_enabled": false
}

###

POST {{baseUrl}}/config HTTP/1.1
Content-Type: application/json

{
  "tesla_enabled": true,
  "tesla_username": "username",
  "tesla_password": "password"
}

###

POST {{baseUrl}}/config HTTP/1.1
Content-Type: application/json

{
  "tesla_vehidx": 0
}
```

or from cURL

```
curl --request POST \
  --url http://openevse-a7d4.lan/config \
  --header 'content-type: application/json' \
  --header 'user-agent: vscode-restclient' \
  --data '{"tesla_enabled": true,"tesla_username": "username","tesla_password": "password"}'
```

commit 6b095af8 (2020-06-26): update docs fix #86
closed 2020-06-26 by glynhudson