OpenEVSE/openevse_esp32_firmware #88: Captive portal is an open WiFi network

DrFrankReade, 2020-06-16:
In 3.2.0 dev:

The captive portal / stand-alone network initializes as an open network - that is, no password or security. Should the wifi connection fail, an open network would present itself to the area.

glynhudson, 2020-06-18:
Are you sure? Are you able to post a screenshot of this? For me 3.2.0 creates an AP protected with the standard password. Is this an issue which just effects 3.2.0?

DrFrankReade, 2020-06-30:
Jesus Glyn, I'm an idiot. I'd flashed an ancient version to that module ages ago just to see if it would compile, and put it away. Then, two weeks ago, I pulled it out of a drawer, flashed 3.2 and was horrified that the phone just logged in. Turns out it had saved the password. Sorry for the scare and my typical amateur mistake.

closed 2020-06-30 by DrFrankReade