OpenEVSE/openevse_esp32_firmware #90: Charging Stuck at 6A

Pukkahq, 2020-06-16:
I have just had our car plugged in with excess solar. the charge rate was stuck at 6A.
I cannot be certin what the charge rate was set to this morning when I disconnected the car. (does it go back to 32A like the old version?)

[image]

[image]

jeremypoulter, 2020-06-17:
With the divert mode off, what is the Max charge rate?

Pukkahq, 2020-06-17:
When i toggled the button it was still 6A, I manually changed it to 32A and then turned eco mode back on

Pukkahq, 2020-06-17:
Here is now, before i plugged the car

[image]

and once the car is plugged in
[image]

Pukkahq, 2020-06-17:
Changing the Max Current to 32A made no difference
![image](https://user-images.githubusercontent.com/15746724/84898234-d4384400-b09e-11ea-9831-03318ce0064d.png

Set the charger to normal and then change Max Current allowed it to go about 6a then back to eco and all is well

[image]

jeremypoulter, 2020-06-18:
Would you be able to send us an export of the grid IE and voltage feeds? I am not quite sure what is going on.

glynhudson, 2020-06-18:
> I cannot be certin what the charge rate was set to this morning when I disconnected the car. (does it go back to 32A like the old version?)

Ecomode is persistent in V3.2.0, so ecomode will stay enabled between charging sessions until it's disabled via the web interface. The timer can be used as a boost timer to charge the EV at full speed at an overnight off-peak time.

Pukkahq, 2020-06-20:
same happened again today. only way to fix was normal mode adjust the Max Current and wait a few seconds for it to take affect. then go back to eco.

Also when in its broken state a reboot of the esp did not fix it.

Pukkahq, 2020-06-22:
[image]

glynhudson, 2020-06-23:
Is it possible to take a screencast video of the OpenEVSE interface to illustrate this issue. I'm struggling to re-create, the new solar PV divert seems to be working ok for me and does not seem to get stuck. It will be interesting to see how the figures change overtime which is not possible to see from a screenshot.

Pukkahq, 2020-06-23:
Would you like access to the charger?

glynhudson, 2020-06-23:
That would be great if you could arrange that. Please contact me privately support@openenergymonitor.zendesk.com

glynhudson, 2020-06-26:
Here is a screencast video of @Pukkahq EVSE in action:

https://www.awesomescreenshot.com/video/275989?key=0532f3e2a01931c520bac3f85d58f3a0

I notice that when the EVSE was paused the excess feed was positive. The Grid IE is expecting positive import / negative export therefore the excess feed should be negative (exporting) when the EVSE is paused. 

Do you have any thoughts @jeremypoulter ?

tentious, 2020-07-30:
I've also noticed the same here. Stuck at 6A.

[image]

Pukkahq, 2020-08-08:
I have just lost connection for a few min when it came back, the last update was 9 min ago.
[image]

Pukkahq, 2020-08-10:
Dont know if this is helpful.

[image]

$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$SV 240760^06
$NK^21
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$SV 242310^07
$NK^21
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GU^36
$OK 5066499 262383^1F
$GF^25
$OK 0 1 0^31
$GE^26
$OK 10 0001^20
$GS^30
$OK 03 2105 03 0540^27
$GG^24
$OK 10120 -1^0E
$GP^33
$OK 462 -2560 -2560^30
$GU^36
$OK 5094645 262383^11
$GF^25
$OK 0 1 0^31
$GE^26
$OK 10 0001^20
$GS^30
$OK 03 2117 03 0540^24
$GG^24
$OK 10120 -1^0E
$GP^33
$OK 462 -2560 -2560^30
$GU^36
$OK 5123243 262383^1E
$GF^25
$OK 0 1 0^31
$GE^26
$OK 10 0001^20

Pukkahq, 2020-08-10:
$GG^24
$OK 9900 -1^3C
$SV 242140^00
$NK^21
$GP^33
$OK 462 -2560 -2560^30
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GU^36
$OK 5151979 262383^19
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GF^25
$OK 0 1 0^31
$GE^26
$OK 10 0001^20
$SV 240130^05
$NK^21
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$SC 9 V^5B
$OK 9^39
$GS^30
$OK 03 2141 03 0540^27
$GE^26
$OK 9 0001^18
$GG^24
$OK 9240 -1^33
$GG^24
$OK 9240 -1^33
$SV 241710^00
$NK^21
$GE^26
$OK 9 0001^18
$GG^24
$OK 9240 -1^33
$SC 10 V^63
$OK 10^01
$GE^26
$OK 10 0001^20
$GG^24
$OK 9240 -1^33
$GP^33
$OK 462 -2560 -2560^30
$GE^26
$OK 10 0001^20
$GG^24
$OK 9020 -1^37
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GU^36
$OK 5179369 262383^18
$SV 242180^0C
$NK^21
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GF^25
$OK 0 1 0^31
$GE^26
$OK 10 0001^20
$GG^24
$OK 9900 -1^3C
$GE^26
$OK 10 0001^20
$GS^30
$OK 03 2153 03 0540^24
$SV 242700^02
$NK^21
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GG^24
$OK 10120 -1^0E
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$GP^33
$OK 462 -2560 -2560^30
$SV 243110^04
$NK^21
$GU^36
$OK 5207747 262383^1A
$GE^26
$OK 10 0001^20
$GG^24
$OK 10120 -1^0E
$SC 8 V^5A
$OK 8^38
$GE^26
$OK 8 0001^19
$GG^24
$OK 10120 -1^0E
$GE^26
$OK 8 0001^19
$GG^24
$OK 10120 -1^0E
$SC 7 V^55
$OK 7^37
$GF^25
$OK 0 1 0^31

glynhudson, 2020-09-08:
Other users have not reported this issue. Unable to replicate

closed 2020-09-08 by glynhudson