OpenEVSE/openevse_esp32_firmware #91: Add LED NeoPixel

chris1howell, 2020-06-18:
Add support for LED NeoPixel.

Development OpenEVSE WiFi integrated 5 - 12v switch mode power supply, two neoPixels and MCP9808 temperature sensor.

Code below successfully tested on development hardware.
 
`
#include <Adafruit_NeoPixel.h>

#define NEO_PIN 17  //ESP32 PIN
#define NEO_LEDS 2 //Number of LEDS

Adafruit_NeoPixel strip = Adafruit_NeoPixel(NEO_LEDS, NEO_PIN, NEO_GRB + NEO_KHZ800);

long previousMillis_NEO = 0;      
long interval_NEO = 10;  
int color = 0;
int state = 0;
boolean fade_in = true;

void setup() {
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
}

void loop() {
  if(state == 0) breathAll(color, color, color); //White
  if(state == 1) breathAll(0, color, 0); //Green
  if(state == 2) breathAll(color, color, 0); //Yellow
  if(state == 3) breathAll(0, color, color); //Blue
  if(state == 4) breathAll(color, 0, 0); //Red
  if(state == 5) breathAll(color, 0, color); //Purple
  
//DEMO breath Each Color
if(state == 6) state = 0;
 if (fade_in == false & color == 1){
  state++;
  fade_in = true;
 }
//End DEMO breath Each Color
}

void breathAll(int r,int g,int b) {
   if(millis() - previousMillis_NEO > interval_NEO) {
    if(fade_in) color++;
    if(!fade_in) color--;
    for(int pix=0; pix<strip.numPixels(); pix++) {
      strip.setPixelColor(pix, strip.Color(r,g,b));
      delay(1);
    }
    strip.show();
      if(color == 254) fade_in = false;
      if(color == 0) fade_in = true;
    previousMillis_NEO = millis();  
  }
}
`

chris1howell, 2020-06-19:
[image]

jeremypoulter, 2020-06-19:
@chris1howell Do you have any thoughts on what the NeoPixels should be used for?

chris1howell, 2020-08-25:
State = 00  "Unknown" WHITE 

State = 01  "Ready" GREEN

State = 02  "Connected" YELLOW

State = 03  "Charging" BLUE
If TEMP > 60 YELLOW

State = FE "Sleeping" PURPLE

Error States RED

State = 04 "VENT REQUIRED"
State = 05 "DIODE CHECK"
State = 06 "GFI FAULT"
State = 07 "NO GROUND"
State = 08 "STUCK RELAY"
State = 09 "SELF TEST"
State = 0A "OVER TEMP"
State = 0B "OVER CURRENT"
State = FF "DISABLED"

chris1howell, 2020-09-18:
Sample code for static color... no animation.

#include <Adafruit_NeoPixel.h>

#define NEO_PIN 17 //ESP32 PIN
#define NEO_LEDS 2 //Number of LEDS

Adafruit_NeoPixel strip = Adafruit_NeoPixel(NEO_LEDS, NEO_PIN, NEO_GRB + NEO_KHZ800);

int color = 0;
int state = 0;

void setup() {
strip.begin();
strip.show(); // Initialize all pixels to 'off'
}

void loop() {
if(state == 0) LEDAll(color, color, color); //White
if(state == 1) LEDAll(0, color, 0); //Green
if(state == 2) LEDAll(color, color, 0); //Yellow
if(state == 3) LEDAll(0, color, color); //Blue
if(state == 4) LEDAll(color, 0, 0); //Red
if(state == 5) LEDAll(color, 0, color); //Purple

void LEDAll(int r,int g,int b) {
for(int pix=0; pix<strip.numPixels(); pix++) {
strip.setPixelColor(pix, strip.Color(r,g,b));
delay(1);
}
strip.show();
}

chris1howell, 2020-11-04:
Hardware v1 Merged. 

Here are the videos for the LEDs on a station without display.

Out of the box, not connected to WiFi or Standalone Mode:
Green - Yellow Flashing - No change with EVSE State
https://drive.google.com/file/d/1E43DihgqMW4t6y9e8wTEIcrrKaJty8mP/view?usp=sharing

Connected Standalone
https://drive.google.com/file/d/1E1R-nqyxMRbVt3hDiMWz1VJTc3sUui0u/view?usp=sharing
Green - Green Flashing  - No change with EVSE State

Connected WiFi
Displays EVSE State
https://drive.google.com/file/d/1DsQXygu4cyQ84rVYbfS2xzGtCzEpcFAq/view?usp=sharing

glynhudson, 2020-11-05:
Nice! Is it possible to have an option to turn off the LED's or reduce the
brightness for discrete operation? I'm sure there will be circumstances
where neighbours would not appreciate a bright light shining into their
garden!

jeremypoulter, 2021-03-07:
Added brightness to the config, just needs GUI. will do that in another ticket though

closed 2021-03-16 by chris1howell