OpenEVSE/openevse_esp32_firmware #92: MCP9808

chris1howell, 2020-06-18:
Add support for MCP9808 temperature sensor.

Development OpenEVSE WiFi integrated 5 - 12v switch mode power supply, two neoPixels and MCP9808 temperature sensor.

Code successfully tested on development hardware.

`
#include <Wire.h>
#include <Adafruit_MCP9808.h>

Adafruit_MCP9808 tempsensor = Adafruit_MCP9808();

void setup() {
  Wire.begin();
}

void loop() {
  float c = tempsensor.readTempC();
}
`

jeremypoulter, 2020-06-19:
Should the temp just be exposed via MQTT/EmonCMS like the other temps, or are there additional requirements?

glynhudson, 2020-07-22:
@chris1howell do we need a RAPI command to pass the temperature reading into the open_evse controller so it can be used for in the normal way for over temperature throttling etc.

chris1howell, 2020-07-22:
It would probably be a good idea to pass the temp to the controller.

What I think we should do long term is have the controller only do the
safety shutdown at a single point 72C (or whatever we set) and have the
WiFi do the throttling.

The WiFi has a lot more resources to fine tune the current and will be
controlling for many other parameters.

commit b8988756 (2020-10-13): Added support for the temp sensor on the on-board temp sensor

Untested as yet as I am not sure my board is working...

Fixes #92
closed 2020-11-04 by chris1howell