OpenEVSE/openevse_esp32_firmware #94: Disable charge until tomorrow while in eco mode

Pukkahq, 2020-06-20:
Disable charge until tomorrow while in eco mode, as we lose the sun it would be good to have a cutoff time or button that stops it trying to charge in eco mode.

jeremypoulter, 2020-06-20:
The new 3.2 build should do this automatically both during dull periods and at the end of the day. Are you not seeing this?

Pukkahq, 2020-06-23:
Hi Jeremy 
It does stop the charge, I am just pondering the idea of eco charge from 8am untill 5pm after that time the solar becomes a little sporadic and might be a way to look after the contactor a liitle. I have my timeout to 5 min as I dont want to discharge the my solar battries. or could we look at lowering the charge to 0A from my testing a few years ago our BMW i3 was happy to go this low.

jeremypoulter, 2020-06-23:
Hi, That is what the smoothing is for, that should filter out all the irregularities for use when controlling the contactor. If you are regularly seeing that there is an issue try tweaking the attack/decay values.

closed 2020-06-23 by glynhudson