OpenEVSE/openevse_esp32_firmware #99: Can't get MQTT to work except for in-built OpenEVSE connection

mikebaz, 2020-07-15:
I'm trying to get MQTT to cooperate on 3.1.0 and 3.2.0 (both have the same issue), and just not having any luck.  The 3.1.0-dev firmware ships with hard-coded defaults to talk to data.openevse.com and those work, but any other connection doesn't seem to work.

For example, setting the protocol to mqtt (no "s" to avoid TLS certificate concerns), the host to test.mosquitto.org, and a blank username and blank password, never connects.  I know that combination for MQTT in general should work because this works:

`mosquitto_pub -h test.mosquitto.org -m "testing" -t test -d`

Not sure what is going on nor how to proceed with troubleshooting, but I'd really like to get this sorted if possible...

mikebaz, 2020-07-15:
also just to add a small bit of detail - one of the MQTT connections I've tried is against a Microsoft Azure IoT Hub.  When I look at the diagnostic logs from that side, it looks like the charger never even attempts to connect to the hub - I don't see a failed authentication event for example.

jeremypoulter, 2020-07-20:
I was able to replicate this with test.mosquitto.org, will do some investigations

glynhudson, 2020-08-04:
What is the conclusion to this? MQTT seems to work fine connecting to a local mosquito server on a Raspberry Pi e.g emonPi

mikebaz, 2020-08-04:
@glynhudson are you connecting by IP or name or?  Just wondering why it's working for you when it's not for me and @jeremypoulter said he was able to reproduce the problem.

glynhudson, 2020-08-05:
I've tested both IP address and hostname and both work for me

jeremypoulter, 2020-08-05:
@glynhudson did you try test.mosquitto.org with no user/pass? I tried that and it did not connect for me. Just have not had a chance to check yet.

glynhudson, 2020-08-06:
Ah your right,  test.mosquitto.org with no user/pass does not work for me either. 

I've also noticed the module requires a reboot after changing MQTT server info to force a reconnect.

jeremypoulter, 2020-08-20:
Tried this again, turns out I had a space at the start of the name, removed the space and all works fine now

jeremypoulter, 2020-08-20:
I have added a few checks to strip spaces from some of the config values where it would not be valid: 
OpenEVSE/openevse_wifi_gui#30

mikebaz, 2020-08-20:
Thank you for looking at this!  At the risk of pushing my luck, is it possible to bubble up some kind of error code or log it or something, so that if the connection fails, there's some indication of how it failed?  That might have caught this kind of thing assuming that's the source of my issue - seeing incorrect whitespace between quotes would have been a big help.

I realize I'm asking a lot so feel free to tell me no and close this :)

glynhudson, 2020-08-24:
> Thank you for looking at this! At the risk of pushing my luck, is it possible to bubble up some kind of error code or log it or something, so that if the connection fails, there's some indication of [...]

We've recently implemented a debug log view via the web interface https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/101

closed 2020-08-24 by glynhudson

mikebaz, 2020-08-28:
OK, this still isn't working for me in 3.3 and the debug log view isn't showing anything related to the MQTT call.  I haven't had a lot of time to dive into it more yet but hope to over the next couple of days to see if I can figure out what's going on.

mikebaz, 2020-08-29:
I can see a connect message coming from the device so I'll keep diving in.  It would be helpful to see the failures in the web-presented log but that's a nice to have.  Thanks for your time getting 3.3 out with changes.  :)

jeremypoulter, 2020-08-29:
What you may need to do is define `ENABLE_DEBUG_MQTT` to get the Debug output from MQTT, I will also try and add failure reports similar to EmonCMS, however MQTT by its nature is not as verbose as HTTP so not sure how useful these will be