OpenEVSE/openevse_esp32_firmware #102: WIFI

Pukkahq, 2020-08-05:
With version 3.3.0 the unit is disconnecting from the wifi and not reconnecting. need to hard reboot the charger. access point is 2 meters away.

Pukkahq, 2020-08-05:
here are the lines from the log from when it was booted until it crashed

OpenEVSE WiFI 7b44
Firmware: 3.3.0
IDF version: 3.30300.190916
Free: 312760
Connecting to SSID: PandC_Inf
Server started
Connected, IP: 192.168.1.216
05-Aug-20, 19:22:50: Free memory 240876 - diff 5372 5372
05-Aug-20, 19:22:52: Free memory 243076 - diff -2200 3172
05-Aug-20, 19:22:54: Free memory 243224 - diff -148 3024
05-Aug-20, 19:22:56: Free memory 242540 - diff 684 3708
05-Aug-20, 19:22:58: Free memory 241548 - diff 992 4700
05-Aug-20, 19:23:00: Free memory 238096 - diff 3452 8152
05-Aug-20, 19:23:04: Free memory 238024 - diff 72 8224
05-Aug-20, 19:23:06: Free memory 237704 - diff 320 8544
05-Aug-20, 19:23:08: Free memory 237816 - diff -112 8432
05-Aug-20, 19:23:10: Free memory 237912 - diff -96 8336
05-Aug-20, 19:23:12: Free memory 237664 - diff 248 8584
05-Aug-20, 19:23:14: Free memory 237660 - diff 4 8588
05-Aug-20, 19:23:16: Free memory 236624 - diff 1036 9624
05-Aug-20, 19:23:18: Free memory 236552 - diff 72 9696
05-Aug-20, 19:23:20: Free memory 236548 - diff 4 9700
05-Aug-20, 19:23:22: Free memory 236512 - diff 36 9736
05-Aug-20, 19:23:24: Free memory 236548 - diff -36 9700
05-Aug-20, 19:23:26: Free memory 236644 - diff -96 9604
05-Aug-20, 19:23:28: Free memory 236200 - diff 444 10048
05-Aug-20, 19:23:30: Free memory 236124 - diff 76 10124
05-Aug-20, 19:23:32: Free memory 236120 - diff 4 10128
05-Aug-20, 19:23:34: Free memory 236460 - diff -340 9788
05-Aug-20, 19:23:36: Free memory 236568 - diff -108 9680
05-Aug-20, 19:23:38: Free memory 236552 - diff 16 9696
05-Aug-20, 19:23:40: Free memory 238060 - diff -1508 8188
05-Aug-20, 19:23:43: Free memory 236120 - diff 1940 10128
05-Aug-20, 19:23:45: Free memory 236084 - diff 36 10164
05-Aug-20, 19:23:47: Free memory 236608 - diff -524 9640
05-Aug-20, 19:23:49: Free memory 230108 - diff 6500 16140
05-Aug-20, 19:23:51: Free memory 235184 - diff -5076 11064
05-Aug-20, 19:23:53: Free memory 235220 - diff -36 11028
05-Aug-20, 19:23:55: Free memory 235180 - diff 40 11068
05-Aug-20, 19:23:57: Free memory 232728 - diff 2452 13520
05-Aug-20, 19:23:59: Free memory 235064 - diff -2336 11184
05-Aug-20, 19:24:01: Free memory 235104 - diff -40 11144
05-Aug-20, 19:24:03: Free memory 235180 - diff -76 11068
05-Aug-20, 19:24:05: Free memory 231972 - diff 3208 14276
05-Aug-20, 19:24:07: Free memory 235180 - diff -3208 11068
05-Aug-20, 19:24:09: Free memory 235108 - diff 72 11140
05-Aug-20, 19:24:13: Free memory 235148 - diff -40 11100
05-Aug-20, 19:24:17: Free memory 235216 - diff -68 11032
05-Aug-20, 19:24:19: Free memory 235144 - diff 72 11104
05-Aug-20, 19:24:21: Free memory 231968 - diff 3176 14280
05-Aug-20, 19:24:23: Free memory 235076 - diff -3108 11172
05-Aug-20, 19:24:25: Free memory 228100 - diff 6976 18148
05-Aug-20, 19:24:27: Free memory 235180 - diff -7080 11068
05-Aug-20, 19:24:29: Free memory 234900 - diff 280 11348
05-Aug-20, 19:24:31: Free memory 232012 - diff 2888 14236
05-Aug-20, 19:24:33: Free memory 231620 - diff 392 14628
05-Aug-20, 19:24:35: Free memory 234752 - diff -3132 11496
05-Aug-20, 19:24:37: Free memory 235144 - diff -392 11104
05-Aug-20, 19:24:39: Free memory 235180 - diff -36 11068
05-Aug-20, 19:24:41: Free memory 235216 - diff -36 11032
05-Aug-20, 19:24:43: Free memory 235144 - diff 72 11104
05-Aug-20, 19:24:45: Free memory 232908 - diff 2236 13340
05-Aug-20, 19:24:47: Free memory 235108 - diff -2200 11140
05-Aug-20, 19:24:49: Free memory 235220 - diff -112 11028
05-Aug-20, 19:24:51: Free memory 234768 - diff 452 11480
05-Aug-20, 19:24:55: Free memory 234696 - diff 72 11552
05-Aug-20, 19:24:57: Free memory 235148 - diff -452 11100
05-Aug-20, 19:24:59: Free memory 231928 - diff 3220 14320
[... 485 lines cut ...]
05-Aug-20, 19:42:46: Free memory 227196 - diff 3988 19052
05-Aug-20, 19:42:48: Free memory 230896 - diff -3700 15352
05-Aug-20, 19:42:50: Free memory 230800 - diff 96 15448
05-Aug-20, 19:42:52: Free memory 230516 - diff 284 15732
05-Aug-20, 19:42:54: Free memory 228968 - diff 1548 17280
05-Aug-20, 19:42:56: Free memory 230484 - diff -1516 15764
05-Aug-20, 19:42:58: Free memory 228892 - diff 1592 17356
05-Aug-20, 19:43:00: Free memory 230408 - diff -1516 15840
05-Aug-20, 19:43:02: Free memory 230484 - diff -76 15764
05-Aug-20, 19:43:04: Free memory 228932 - diff 1552 17316
05-Aug-20, 19:43:06: Free memory 229384 - diff -452 16864
05-Aug-20, 19:43:08: Free memory 230964 - diff -1580 15284
05-Aug-20, 19:43:10: Free memory 230892 - diff 72 15356
05-Aug-20, 19:43:12: Free memory 230432 - diff 460 15816
05-Aug-20, 19:43:14: Free memory 230472 - diff -40 15776
05-Aug-20, 19:43:16: Free memory 222804 - diff 7668 23444
05-Aug-20, 19:43:18: Free memory 230752 - diff -7948 15496
05-Aug-20, 19:43:20: Free memory 230716 - diff 36 15532
05-Aug-20, 19:43:22: Free memory 230644 - diff 72 15604
05-Aug-20, 19:43:24: Free memory 230608 - diff 36 15640
05-Aug-20, 19:43:26: Free memory 230644 - diff -36 15604
05-Aug-20, 19:43:28: Free memory 229584 - diff 1060 16664
05-Aug-20, 19:43:30: Free memory 230552 - diff -968 15696
05-Aug-20, 19:43:32: Free memory 230480 - diff 72 15768
05-Aug-20, 19:43:34: Free memory 228760 - diff 1720 17488

Pukkahq, 2020-08-05:
ess:^47
$OK^20
$FP 11 0       ^22
$OK^20
$FP 0 1 192.168.1.216^3C
$OK^20
$FP 13 1       ^21
$OK^20
$GU^36
$OK 0 226594^1E
$SV 236590^0A
$NK^21
$GE^26
$OK 15 0001^25
$GG^24
$OK 0 -1^0C
$GE^26
$OK 15 0001^25
$GC^20
$OK 6 32 15 15^17
$GE^26
$OK 15 0001^25
$G3^50
$OK 0^30
$GH^2B
$OK 0^30
$GE^26
$OK 15 0001^25
$GE^26
$OK 15 0001^25
$GE^26
$OK 15 0001^25
$GE^26
$OK 15 0001^25
$GE^26
$OK 15 0001^25
$GE^26
$OK 15 0001^25
$FF T 1^41
$OK^20
$GE^26
$OK 15 0001^25
$GD^27
$OK 0 0 0 0^20
$GF^25
$OK 0 0 0^30
$GE^26
$OK 15 0001^25
$GG^24
$OK 0 -1^0C
$GE^26
$OK 15 0001^25
$F0 1^43
$OK^20
$SV 237250^00
$NK^21
$GS^30
$OK fe 0 00 0700^14
$GE^26
$OK 15 0001^25
$GG^24
$OK 0 -1^0C
$GG^24
$OK 0 -1^0C
$GP^33
$OK 342 -2560 -2560^35
$SV 237390^0D
$NK^21
$GE^26
$OK 15 0001^25
[... 3998 lines cut ...]
$OK 32 0001^20
$GG^24
$OK 0 -1^0C
$GE^26
$OK 32 0001^20
$GG^24
$OK 0 -1^0C
$GE^26
$OK 32 0001^20
$GS^30
$OK fe 21 03 0500^26
$GG^24
$OK 0 -1^0C
$SV 236670^07
$NK^21
$GE^26
$OK 32 0001^20
$GG^24
$OK 0 -1^0C
$GP^33
$OK 335 -2560 -2560^35
$GE^26
$OK 32 0001^20
$GG^24
$OK 0 -1^0C

glynhudson, 2020-08-06:
I have shortened and hidden the log output since it's not relevant to this issue and made the post hard to read. 

- Are you using the standard ESP32 Huzzah module 
- And standard OpenEVSE power supply and controller? 
- What is the RSSI of your WiFi network as reported by the OpenEVSE? 

I've been unable to replicate the disconnection issue. We will investigate further?

Pukkahq, 2020-08-08:
Its just stopped. 
Yes standard PSU and controller from the shop.
Wifi is -72 

Went back to 3.2 for a few days and the wifi was ok.

here is the last part of the log
$OK 6 0001^17
$GS^30
$OK 03 11966 03 0540^18
$GG^24
$OK 6160 -1^3D
$SV 243200^06
$NK^21
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$SV 242820^0F
$NK^21
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$SV 244890^02
$NK^21
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$SV 243390^0E
$NK^21
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$SV 243690^0B
$NK^21
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$GE^26
$OK 6 0001^17
$GG^24
$OK 5940 -1^34
$SC 7 V^55
$OK 7^37

glynhudson, 2020-09-08:
Unable to replicate

closed 2020-09-08 by glynhudson