OpenEVSE/openevse_esp32_firmware #106: Solar PV divert should accept inverted sign export information

treepleks, 2020-08-05:
My Inverter generates excess production information with the opposite sign:

- Import is negative
- Export is positive

It would be nice to have this as a 3rd possibility.

glynhudson, 2020-08-06:
I guess we could add an option to invert the sign via the OpenEVSE WiFi interface. However, this could confuse some users. It think it would be easier in your circumstance to use nodeRED to invert sign and then post to a new MQTT topic so the sign is the correct wya round for the OpenEVSE: Positive Import / Negative Export.

treepleks, 2020-08-07:
I was thinking of adding a third item in the "Feed" selection list in the Solar PV divert panel that already includes ""Grid (+I/-E)". This could be renamed to "Grid Import (+I/-E)" and a new "Grid Export (-I/+E)" item could be added.

On my side, I already solved this using a python/mqtt script that publishes the negated wattage. But I would assume that not all people will be happy doing this.

Pukkahq, 2020-08-07:
I have the same problem, I dont have node red installed so i just -feed -feed which gives the same result

glynhudson, 2020-08-07:
Ok sure, we'll add an option to the interface invert the sign

closed 2026-09-21 by github-actions[bot]