OpenEVSE/openevse_esp32_firmware #108: Minimum charge time

Pukkahq, 2020-08-08:
I have just been sat watching the charger and thinking about how this setting works. today we have had a lot of sun then some clouds.

Does this setting mean that one the charge rate is <6A for minimum charge time it turns off. 

Or

Once the charging has started and minimum charge time has been reached (300 sec in my case) and <6A stop charging.

jeremypoulter, 2020-08-09:
Is the second one, after the charge has started to the min time had to have passed  before the charge will stop. Covers a few edge cases where just the smoothing still caused a lot of transitions.

Pukkahq, 2020-08-10:
That makes more sense on what i was seeing over the weekend, we had sun then a little cloud then more sun and I was trying to work out why it was turning off after a few moments of cloud.

At the moment we have 300 seconds set. is it possible for the 300 seconds to start when the charge rate <6A

closed 2020-08-10 by glynhudson

Pukkahq, 2020-08-10:
Can I as why this has been closed?

glynhudson, 2020-08-11:
Has Jeremy not answered your questions? I can't see an issue here

Pukkahq, 2020-08-11:
He did answer the question, I feel that an improvement to the timer and it should start once the solar is <6A. this would allow for the cloudy days we have in the UK. sometimes its sunny for 20 min amd producing 6000 watts and then you have a short period of 3-4 min of cloud then back to bright sun.

jeremypoulter, 2020-08-13:
That is basically what the smoothing is for. The smoothing worked out better in a lot more cases than static timers, so to get a similar effect you can just decrease the decay value.

Pukkahq, 2020-08-13:
Many thanks for that information, Maybe a little more infomations is need on how the smoothing works as i just keep changing them.

mikeh2010, 2020-08-19:
Yes, I would like to understand the thinking behind the smoothing attack, decay and minimum charge time.