OpenEVSE/openevse_esp32_firmware #109: Improving Divert charge start power given electricity buy/sell prices

treepleks, 2020-08-14:
Currently, in the Grid export divert mode, I understand, reading the code, that OpenENVSE tries to eat almost all the available export power (GridE, positive when export) for charging (with a reserve of 100W), which looks nice. 

However, it only starts charging if the available power is greater than minP = MinA * GridVoltage (where MinA is the minimum charge of 6A, GridVoltage  is the current grid voltage and MinP is therefore the minimum charging power).

The current policy is not optimal in terms of costs: in most cases, charge should start before the minimum charging power (minP) is available. Let me explain why.

In my country, I resell solar electricity to my energy provider for R units of costs / KWh.
I also buy the KWh at D units of cost / kWh during day. Alternatively, I can charge my car at night (or at another low-cost energy source) for N units of costs / KWh.

Usually R < N <= D (that's my case).

Notice that if my current GridE is slightly below minP but almost equal to minP, I'm selling minP kW at R units / hours.
If I were charging instead, I would store electricity which I would otherwise need to buy at D units of cost /kWh now (or N only, later at night or from another cheap source not available now). So, I'm actually losing money (if R < N).

So, when should we start charging?

Well as soon as the remaining power (minP - GridE) we need to charge now (instead of at night) costs us less than what using the free solar power now would bring if charging (we would sell it for R but charging means we instead store something that would later cost at least N later). So...

       (Pmin-GridE)*(D-N) < GridE*(N-R) 
                    <=> 
        GridE > Pmin*(D-N)/(D-R)

In my case R= 0.065, D = 0.1779, N = 0.1335, so assuming that MinA = 6A and GridVoltage = 240V, we have Pmin = 1440W and we should start charging when GridE > 1440*(0.0444/0.1129) = 566 W. Yes, as soon as I produce 544 extra watts, I should decide to eat them all and consume an extra 896 (costly) W (that would anyway cost at least 0.1335 anyway).

NB: this formula also tells us that if you do not have access to electricity at costs lower than D and given that R < D, it's better to charge as soon as some gridE is available and the formula becomes GridE > 0.

NB2: the actual behavior corresponds to R = N (we resell at the same price as the minimum cost). This could be set as a default for continuity of behavior.

NB3: I would be happy if someone could reread this. It looks good to me but a second eye/brain is always better!

Would it be possible to change the code/wifi interface according to this? The wifi interface would ask for R (selling price per kWh) D (buying price during day) and N (minimum usable buying price).

Also, this means the idea of having a reserve of 100W is also suboptimal (when R < N < D). Essentially, we lose money not using these 100W now. We sell them R but storing them in the car instead would replace something that would cost N (a least). Since we can control the current by +/- 1A or the power by power increments of GridVoltage Watts. We should strive to have a negative GridE of -Voltage.(1-(D-N)/(D-R)) (in y case -240*(1-(0.0444/0.1129)) = -146W). So, only decrease the charging power when gridE < -146W (instead of trying to have it > 100W)

If you'd like a PR for this, I can try to write this down in C++ but it may take some time :-)

slavkobrzeg, 2020-09-02:
You're right! It is always better to use 100% of produced energy rather than give it back (or sell it) to the vendor and then take it (buy it) again. In my country an electricity provider takes 20% of my energy as a fee for storing my energy. So it is probably clearer to understand that I lost 20% of every kWh I push to the grid. We can translate it to the money ground by saying that they "pay" me 20% less for my energy than I pay for theirs - R = D * 0.80. 
So, as much as I can keep it and use it on my side the better for me. Of course sometimes it is not possible when we do not have any energy buffer like a car to charge or a water tank to heat...
In other words it is better to even buy a bit of energy from the grid if that would mean that we can utilize 100% of our own energy.
I think though, that instead of taking your very sophisticated algorithm it would be easier to be able to edit that GridE (Grid Export) value above which a charging starts, but of course your algorithm will be better even if many times D = N.
I vote for that change.

glynhudson, 2020-09-08:
I understand your point, however most users seem to prefer all the power to be provided by the PV when charging on Eco mode i.e EVSE waits for solar PV or excess to be  > 1.4kW before starting to charge. For most domestic PV systems 1.4kW is not a particularly high power level and is often reached by mid-morning. 

If you prefer the unit to switch on sooner you just need to make a MQTT feed with the solar PV / grid excess value multiplied by a scale value then use the scaled MQTT feed in the EVSE. NodeRED is useful tool to do this.

closed 2020-09-08 by glynhudson

treepleks, 2020-09-10:
Dear Glyn,

You are in charge of course, so I will just be kindly trying to change your mind here (with good arguments I think :-) ). I really do think there is an easy and attractive way to offer this functionality to the whole public of OpenEVSE. One which is also grounded into facts. As Einstein said, "It's usually harder to have someone change his decision than breaking an atom" but I still have faith in the power of facts.

1) I have already an MQTT python script running to generate inverted power messages to Open EVSE so this would be easy. But the MQTT with either shifted or scaled exported power is not suitable here. If I shift my current power by let say 1000W to start the charge sooner (will start around 400W instead of 1400W), then when/if my truly exported power reaches 1400W, the power used for charging by OpenEVSE will be instead around 2400, with 1000 extra grid Watts that are needlessly used here: the optimal consumption at this point is 6A (~ 1400W) with no grid import. The issue is that there is no simple shifting/scaling rule that will work because the policy to apply depends on the actual wattage sent to the car by Open EVSE. So, really, the best place to do this is inside the Open ENVSE microcontroller. Of course, a more complex setting using RAPI is likely feasible but painful.

2) 1400W reached quickly: this obviously depends on the peak wattage of your installation. In Europe, several countries have subsidized installations of up to 3kW making this level of power very usual. Then, 1400W are not always reached or only for a short duration (depending on how light clouds are). Some OpenEVSE users also have an automated house system that will decide to divert a fraction of the power to heat their water or power their AC system or the pump of their swimming pool, leaving less to the car. This %age of users is probably low (I'm there) but it will increase.

3) Most users prefer all the power to be provided by the PV when charging: I can imagine that even if it's economically suboptimal in almost all cases (as soon as the selling price of electricity is less than the buying prices). However, as I mentioned in my original message, there is a choice of values for the parameters governing this controller that precisely preserve this behavior. It suffices to set these values as "default" values so users will not be prevented nor even surprised by a change of policy: there will be none under the default setting. I also see that slavkobrzeg agrees with me. I assume that most Open EVSE users just don't read Open EVSE GitHub issues, so having just one favorable comment probably hides many more.

Then, I have thought a bit more about this and I think my original mail above and my proposal of interface is way too complex. I assume, possibly wrongly, this may be the main reason for your decision. I would instead propose to just offer a simple parameter on the web interface, one that varies from conservative (1.0) to aggressive (0.0). The conservative policy would be the default setting and would precisely match the current OpenEVSE policy: use only PV watts. Aggressive (0.0) would start using PV watts as soon as available (as slavkobrzeg mentions above). Intermediary values would vary between these two extreme behaviors.

The optimal value to choose (from a pure economical point of view) would be easy to compute: it's just (D-N)/(D-R) in my initial terminology. So, it's easy to compute and set (from RAPI too ideally) and for those that don't want to do maths, it's easy to understand and set "roughly" by hand/experience/feedback.

If this looks better to you, I would even be ready to fork Open EVSE and program this (slowly, I have more projects). However, I would rather not make this effort if the likelihood that this gets integrated into OpenEVSE is close to zero. Even if you give me a green light, I would understand a final rejection, of course, in case of bugs or unexpected behavior or poor code quality. But please, try to kindly save my time by being frank (you can PM me at treepleks@free.fr).

Kind regards,

T. Reepleks