OpenEVSE/openevse_esp32_firmware #112: Add UI to enable/disable front button.

chris1howell, 2020-08-26:
There is currently no way to enable/disable the front button short of unplugging. Stations installed outdoors or in publicly accessible locations are venerable to tampering of settings or interrupting a charge.

RAPI command has been incorporated into openevse to enable/disable button 
$FF B 0
$FF B 1

https://github.com/lincomatic/open_evse/commit/c226ad80166ba79d47b27a96e911276b0a0cacb6

jeremypoulter, 2021-02-18:
With the v4 changes the WiFI module now always disables the button and handles the button presses it the WiFi module, so is there still a requirement to implement the disable button logic?

jeremypoulter, 2021-03-25:
@chris1howell Any thoughts?

chris1howell, 2021-03-25:
Some users want the button disabled regardless of where it is interpenetrated. If someone walks up to an outdoor station they can mess with your station. There should be a way to lockout public access to the button.

closed 2026-09-21 by github-actions[bot]