OpenEVSE/openevse_esp32_firmware #115: Enhancement - PlugShare

chris1howell, 2020-09-12:
Service add-on to give OpenEVSE station owners the ability to share their station and post real-time status.

https://developer.plugshare.com/docs/#station

https://developer.plugshare.com/docs/#station-availability

pir8radio, 2020-09-12:
As I’m looking into this most of their API commands are GETs to retrieve info from them.  I have reached out to PlugShare to gather more info in charger integration.

jeremypoulter, 2020-09-24:
I had a quick look at this, is this just about sharing the charger status, or is there more to it?

pir8radio, 2020-09-24:
> I had a quick look at this, is this just about sharing the charger status, or is there more to it?

So, i think it "could" be more..   PlugShare also does Credit card billing, and I'm working with PlugShare to see if there is some comparability with their API to be able to remotely enable and disable an OpenEVSE charger... (they do it now linking to other networks api's)      Initially I would love to get the charger status online with PlugShare..   But if PlugShare and OpenEVSE could be setup to work together with their full API so that they could monitor status, receive kwh, and enable/disable an openEVSE charger, it would be an opportunity for small businesses to offer people pay charging using openEVSE hardware.  Businesses  that dont have big "charger network" money to buy THEIR chargers and pay their fees.      For example a chargepoint outdoor charger is like $7-12k US for the device, then you have to pay about $3k us every 5 years for them to handle billing and the cellular connection..   There is no way anyone can make money from having a pay charger, without charging the EV user a crazy amount per charge.   Plus it makes it hard for the little guys, or even bigger companies, to cost justify those kind of fees and hardware price.    Having the ability to throw relatively inexpensive OpenEVSE chargers on a free (or minimal cost) network like PlugShare, let PlugShare do the billing, we use our own wifi or cellular.  I could see OpenEVSE outdoor chargers pop up all over the place!

pir8radio, 2022-02-12:
Kind of died out huh?  :-)

varian-ma, 2022-03-12:
> Kind of died out huh? :-)

What is the story here, How far did this go?

pir8radio, 2022-03-12:
I think they support some form of OCPP and said "good enough, figure it out"..  :-)

jeremypoulter, 2022-03-12:
No, just this is one of many features that need implementing, there are a few more architecture changes to be made before we get to this.

This is also an open source project and PRs are welcome.

glynhudson, 2023-03-31:
I think it's a poor show that Plugshare is mandating a propriety API for this, OCPP is made for this and is already implemented and supported by most charging networks around the world.

pir8radio, 2026-08-28:
went nowhere I guess