OpenEVSE/openevse_esp32_firmware #119: Can't use Debug Mode. Failed to launch GDB: .pioinit:15

Mik-el, 2020-09-29:
Hi,
I have an ESP32 board (Espressif ESP32-WROOM-32 [https://i.imgur.com/Sj7URTJ.jpg](url) ) and I'm using Visual Studio Code (v.1.49.2 with espressifIDF v0.5.1).
I downloaded the code in this repo and I successfully flashed it on the esp (I get the wifi access point and the web interface).
I want to use the micro-usb port for Debug Mode but I can't.
According to the readme in this repo I changed all "DDEBUG_PORT" entry in the platformio.ini file to "DDEBUG_PORT=Serial".
When I try to use one of the debug button ("Run->PIO Debug" or "Run->PIO Debug skip pre-debug" or "PlatformIO->Debug->Start Debugging") I get an error window that reports:
`Failed to launch GDB: .pioinit:15: Error in sourced command file: Remote comunication error.Target disconnected.: No error. (from interpreter-exec console "source.pioinit") `

On the Debug Console I get:

> Open On-Chip Debugger  v0.10.0-esp32-20190708 (2019-07-08-11:04)
> Licensed under GNU GPL v2 For bug reports, read http://openocd.org/doc/doxygen/bugs.html none separate
> adapter speed: 20000 kHz
> Info : Configured 2 cores
> esp32 interrupt mask on
> Info : tcl server disabled
> Info : telnet server disabled
> Error: no device found
> Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
> Warn : Flash driver of esp32.flash does not support free_driver_priv()
> Warn : Flash driver of irom does not support free_driver_priv()
> Warn : Flash driver of drom does not support free_driver_priv()
> .pioinit:15: Error in sourced command file:
> Remote communication error.  Target disconnected.: No error.

while in the Problems tab I get:

> c_cpp_properties.json -> Cant find:
> "c:\Users\mikel\Downloads\OPEN_EVSE\GIT_CLONE\ESP32_WiFi_V3.x\include".
> "C:\Users\mikel\.platformio\packages\framework-espidf\components\expat\expat\lib".
> "C:\Users\mikel\.platformio\packages\framework-espidf\components\include_compat".
>  "c:\Users\mikel\Downloads\OPEN_EVSE\GIT_CLONE\ESP32_WiFi_V3.x\include".
> "C:\Users\michele.carnevale\.platformio\packages\framework-espidf\components\expat\expat\lib".
> "C:\Users\mikel\.platformio\packages\framework-espidf\components\include_compat".
> 
> time_man.cpp ->
> 'double diff_time(timeval, timeval)' defined but not used [-Wunused-function]

How can I solve? Thanks in advance

jeremypoulter, 2020-09-29:
That is the board I use for most of my dev so should work no problem. Are you using the correct env? Should be `openevse_esp-wrover-kit_idf`, ie:

```
pio run -e openevse_esp-wrover-kit_idf
```

Mik-el, 2020-09-30:
Thank you Jeremy,
I didn't think about this, I edited the platformio.ini file, I changed `default_envs = openevse_huzzah32_idf` to `default_envs = openevse_huzzah32_idf` but it seems it gave other problems, so I undid this modify and I used your command, it works.

Now I want to debug,
according to the readme.md I have to use the `openevse_dev` env and the command to use is `pio run -t upload -e openevse_dev` 
Considering that I'm using the `openevse_esp-wrover-kit_idf` env which is the correct command to give? I tried `pio run -t upload -e openevse_esp-wrover-kit_idf -e openevse_dev` but it doesn't work for both envs.

jeremypoulter, 2020-09-30:
So to upload you would use `pio run -e openevse_esp-wrover-kit_idf -t upload`. The Readme probably needs updating, that info is still mostly from the ESP8266 build

Mik-el, 2020-10-02:
Thank you.
Yes the readme needs to be updated, there's also a typing error in the debug command `pio run -t upload -eopenevse_dev` (lack of space after -e).

Back to my case, setting the right environment and using the correct debug command you solved my `GDB: .pioinit:15` error. There's just one problem still left, it is:

> Error: no device found
> Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
> ** OpenOCD init failed **
> shutdown command invoked
> 
> Warn : Flash driver of esp32.flash does not support free_driver_priv()
> Warn : Flash driver of irom does not support free_driver_priv()
> Warn : Flash driver of drom does not support free_driver_priv()
> *** [upload] Error 1

I checked in driver management the COM driver for my board. In the "Hardware IDs" tab I don't have vid 0403 and pid 6010 but instead vid 10C4 and pid EA60.
I updated the driver but nothing changes, and according to some people that fixed this error I also downloaded and used "Zadig" to replace my driver with a "WinUSB" driver and "libusbK" driver, but vid and pid IDs don't change...

I think this is the last thing to solve and then I'l lfinally able to debug and close this issue!

Mik-el, 2020-10-19:
Ok I think I solved.
The correct environment that solved my upload problem is **openevse_huzzah32_dev**.
To do a simple debug you need to use "Monitor.
Otherwise if you want to fully debug software on your esp32 (and thus solving the **GDB: .pioinit:15** error) it seems that additional hardware is required (https://www.hackster.io/brian-lough/use-the-platformio-debugger-on-the-esp32-using-an-esp-prog-f633b6)

closed 2020-10-19 by Mik-el